Gissio/radpro #2: Different FS2011

mjwaxios, 2023-06-28:
Just received a FS2011 from amazon (6/27) and the board revision is the same as your picture, Z27809 but the CPU different:
GD32F150
C8T6
CEHH213
AJ2151
GigaDevice
       ARM

After programming, no display, no sounds, buttons don't seem to do anything.  Have not had time to recompile for the different processor yet.

Pictures to follow later...

Gissio, 2023-06-28:
I am minutes away of publish a release that fixes this. Hang tight!

closed 2023-06-28 by Gissio

mjwaxios, 2023-06-28:
Fantastic work.  Works like a champ.   Thanks.

Gissio, 2023-06-28:
I have just uploaded the official 1.2.2 release. Hope you enjoy it!

pixitha, 2023-07-02:
[image]

I have this same CPU type on mine, adding a picture in case you want to add it to the install doc. 1.2.2 works on this CPU perfectly.

Gissio, 2023-07-02:
Nice! Thanks for posting.

That seems to be an M4011 tube. BTW, some of those are photosensitive, so you might want to expose your tube to sunlight and see if the radiation level goes up. If it does, wrap your tube in black plastic.

pixitha, 2023-07-02:
Would it be beneficial to upload the orig backup firmware somewhere for others?

Gissio, 2023-07-02:
From a legal point of view, I don't think that is a good idea.

BTW, did you get a private email?

mjwaxios, 2023-07-05:
> EXTERNAL EMAIL - This email originated from outside of CACI. Do not click any links or attachments unless you recognize and trust the sender.

So my first backup failed (my problem) but I got a second one and pulled the firmware for it.  It would be nice to have a recovery file.  Do you have a copy of it  or do you need a copy (which I now have and tested)?

  *   Michael

This electronic message contains information from CACI International Inc or subsidiary companies, which may be company sensitive, proprietary, privileged or otherwise protected from disclosure. The information is intended to be used solely by the recipient(s) named above. If you are not an intended recipient, be aware that any review, disclosure, copying, distribution or use of this transmission or its contents is prohibited. If you have received this transmission in error, please notify the sender immediately.

Gissio, 2023-07-09:
I have just released Rad Pro 1.3.0 which adds support for the J305 tube.

Gissio, 2023-11-16:
Rad Pro 2.0 Beta 11 is out!

https://github.com/Gissio/radpro/releases