Gissio/radpro #3: Different FS2011

felixsteghofer, 2023-07-10:
Hi,
I received a FS2011 from aliexpress and the board revision seems to be slightly different (e.g. buzzer, tube location + some components), plus the cpu is different
GD32F103
C8T6
CDU1310
DJ2216
GigaDevice
ARM

After programming, no display, no sounds, buttons don't seem to do anything. After writing the backup two or three times it is working again.
[image]

[image]

[image]

[image]

Not really experienced with ARM CPUs so some hints on where/how I can find the correct board definitions would be very beneficial, otherwise I will try on my own, probably in the next days.

Gissio, 2023-07-10:
Interesting. Please go to this project's Security Tab, draft a new security advisory, and send me your email address so I can contact you. Your email address won't be released publicly.

Gissio, 2023-08-06:
Fixed in the upcoming release (1.4.0)!

closed 2023-08-06 by Gissio

Gissio, 2023-11-16:
Rad Pro 2.0 Beta 11 is out!

https://github.com/Gissio/radpro/releases