Gissio/radpro #5: FS9000 with Texas Instruments MSP430F147 MCU

maxromanovsky, 2023-08-06:
FS9000 Dosimeter I have on hand was bought during spring 2022 on AliExpress.
To my surprise, it doesn't have an ST ARM CPU, but a TI MSP430F147 MCU:
https://www.ti.com/product/MSP430F147

Also, it doesn't have a 4-pin XS1 header, but 10-pin J3 header instead.
I've unsoldered the BZ1 buzzer (because stock firmware is buggy, and signals even if alert conditions were not met).
But FT232 & mini-USB connectors were unpopulated.

I guess support for it is out of question, isn't it?
[image]
[image]

Gissio, 2023-08-06:
Cześć!

I have never worked with the MSP430 so I can't support it.

But I can give you some hints as to what would need to be done (general hints, not support):

* Follow the traces from the microprocessor to the board's devices (keyboard, display, battery, etc) and see what port connects to what device. A multimeter might help.
* The 10-pin header is a JTAG connector. Get a JTAG USB adapter, and attempt to download the firmware through it. Then use some MSP430 disassembly tool to disassemble the firmware, and see how the code accesses the MSP430F147 internal devices (input/output ports, timers, ADCs, etc).
* Then, follow the steps in https://github.com/Gissio/radpro/blob/main/docs/developers.md and customize the code for the MSP430/FS9000.

This should make a nice project...

closed 2023-08-06 by Gissio

maxromanovsky, 2023-08-07:
Gotcha, thanks.
While this sounds interesting, I'll skip it for now due to total lack of free time.
Maybe later, who knows?
Anyways, thanks for your response!

eriedj, 2023-08-20:
[image]
pudibei NR-950 bought on ebay, looks to be same mcu and firmware but diff pcb. i may try to take a look into it sometime, but i will note it's been a decade since i last did any real coding.