Gissio/radpro #6: FS1000 USB

murkle, 2023-09-02:
Hi, this looks very cool!

For the FS1000 do you think the direct logging to the computer would work with the built in USB socket?

mressl-itba, 2023-09-02:
Hi there!

Could you attach photos of the electronics board? (If possible, right from above, without light reflections)

Also, could you go to the "Security" tab and report a vulnerability with your email address so I can contact you in private? Your email address won't be published.

murkle, 2023-09-03:
I'm michael at geogebra.org

I don't have a device (yet 😆)

I'm interested in connecting via eg WebSerial or WebHID, will that be possible? I can code the JavaScript side

mressl-itba, 2023-09-04:
First, let me take my hat off for your work at geogebra.org.

I don't own a Bosean FS1000 either, so I can't confirm whether the factory firmware supports live data logging. But, if you are willing to help, it shouldn't be too hard to adapt Rad Pro for the FS1000.

murkle, 2023-09-04:
Thanks!

Shall we take this to email?

Gissio, 2023-11-16:
Rad Pro 2.0 Beta 11 is out!

https://github.com/Gissio/radpro/releases

Gissio, 2024-02-17:
Rad Pro 2.0 Beta 14 is out!

https://github.com/Gissio/radpro/releases

closed 2024-03-13 by Gissio