Gissio/radpro #8: FNIRSI GC-01 support?

cromagn, 2023-09-25:
Hi, _super super_ cool project. 
Is there any possibility of having support for the **FNIRSI GC-01** ? It seems very similar to other devices already supported.
Ciao

Gissio, 2023-09-25:
Salve!

Please go to the "Security" tab, click on "Report a vulnerability" and send me your email address so I can contact you. Your email address won't be published.

cromagn, 2023-09-26:
Done !

Tig3rch3n, 2024-02-04:
Any estimation until the Firmware for the GC-01 is ready?
I just compiled it, but if I try to run it, the Display stays dark and the LED is permanently on.
But I have a copy of the Firmware so I got it back to the Factory Settings

Gissio, 2024-02-16:
I just noticed I overlooked this message.

Good news: now!

Gissio, 2024-02-17:
I just noticed I overlooked this message.

Good news: now! https://github.com/Gissio/radpro/releases

Tig3rch3n, 2024-02-17:
XD I'll be home in 30-40min
I'll try it out tonight

Gissio, 2024-02-22:
I have added a new release. Can you try again?

Tig3rch3n, 2024-02-27:
I'll continue with the others in Debug :)

Gissio, 2024-03-13:
Rad Pro 2.0beta18 was just released!

closed 2024-03-13 by Gissio