Gissio/radpro #9: FS-600

mindlight73, 2023-10-13:
Do you still plan to support FS-600 and if so, when in time?
I just got a FS-600 and it bugs me that there is a USB-C port but the device lacks capability of reporting anything through it.

If you need pictures of the internals, please let me know.

Gissio, 2023-10-13:
FS-600 support is around the corner. Hang tight!

Gissio, 2023-11-16:
Rad Pro 2.0 Beta 11 is out!

https://github.com/Gissio/radpro/releases

Gissio, 2023-11-17:
I forgot to mention. It has support for the Bosean FS-600. Were you able to try it?

mindlight73, 2023-11-17:
Nope. I saw it today. I don't have the serial USB thingy to flash it so I will purchase that tonight and I will open it up and do some soldering next week.... Pray for me 😁

Gissio, 2024-02-17:
Rad Pro 2.0 Beta 14 is out!

https://github.com/Gissio/radpro/releases

mindlight73, 2024-10-19:
Just wanted to add that I finally got a STM32. Yeah, it took a year...
I just flashed my Bosean FS-600. Worked like a charm!

Gissio, 2024-10-19:
I'm glad it worked!