Gissio/radpro #11: Bosean FS-5000 support

ihrapsa, 2023-11-18:
Hi,

I'd like to help getting support for this relatively new [Bosean FS-5000](https://www.bosean.net/FS-5000-Nuclear-Radiation-Detector.html?lang=en&class1=2&searchtype&search=search) Geiger counter. 
Here are some specs:

**MCU:** [ARM Cortex-M4 STM32L431RCT6](https://datasheet.lcsc.com/lcsc/1810010516_STMicroelectronics-STM32L431RCT6_C92468.pdf) 80 MHz, 64KB SRAM, 256KB Flash
**Extra Flash:** winbond 25q64jvsiq - 8 MB serial flash
**USB to TTL:** WCH 340EB27 (basically a CH340)

<details>
  <summary>Click for extra specs: </summary>

| Sensor | XRHH2022-12 Glass Geiger Tube |
|----| ----|
| LCD screen | 2.4-inch TFT LCD |
Measurement range of dose rate | 0.01 uSv/h~50 mSwh
Dose measurement range | 0 uSv~1000Sv
Sensitivity | ＞1 CPS/uSv/h
Energy response | 48keV~1.5MeV
Relative error | ＜10%(1mSv/h)
Power supply | 3.7V rechargeable lithium battery, 1800 mAh
Power consumption | ＜15 mwW (long endurance mode: ≥30 days)
Temperature | -10℃~+45℃
Dimension | 127*65*27(mm)
Weight | 160g (including battery)
</details>

<details>
  <summary>Click for pics: </summary>

[image]  
[image]
[image]
[image]
[image]

</details>

It's also got the ability to connect to a pc to analyse the logged history using their [Windows app](https://www.bosean.net/wp-content/uploads/2023/06/Personal-Radiation-Dose-Management-System.zip)

Output of `dmesg` when connecting to a Linux computer:
```[ 2522.162423] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[ 2522.268542] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[ 2522.268568] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 2522.268585] usb 1-1.3: Product: USB Serial
[ 2522.273344] ch341 1-1.3:1.0: ch341-uart converter detected
[ 2522.278803] usb 1-1.3: ch341-uart converter now attached to ttyUSB0
```
Let me know if there's any other info I should provide 😃

ihrapsa, 2023-11-19:
Just managed to dump the STM32L431RCT6 256kB Flash. 
[Bosean_FS-5000_STLink_Utility_dump.bin.zip](https://github.com/Gissio/radpro/files/13404427/Bosean_FS-5000_STLink_Utility_dump.bin.zip)

Will try to dump the 8MB SPI NOR Flash as well although I'm not sure that's necessary.

Jonah

ihrapsa, 2023-12-11:
ping. :)

Gissio, 2023-12-12:
Thanks for your submissions.

Once I find the time, I'll take a look.

wnasich, 2024-03-26:
@ihrapsa  did you manage to extract data from FS-5000 on Linux? I would like to have an script to extract data online

ihrapsa, 2024-03-27:
Hi, 
At that time I used their windows app to dump the logs or log the live data through a com port over USB. I was thinking about about sniffing the serial communication so we can reverse engineer it. I was looking into `com0com` and `Termite` for port forwarding and sniffing communication. Would this be helpful, is there a better approach?

Jonah

wnasich, 2024-03-27:
@ihrapsa provided that protocol is not availab, sniffing the serial communication is a good approach to me

Red-Owl, 2024-04-03:
this is my test to connect fs5000 to linux on serial port, incomplete.

connect -> 115200, DTR OFF, 8N1

device id? - > AA 04 06 b4 55

AA 05 0E 01 BE 55 -> PC
AA 05 0E 01 BE 55 -> PC
AA 05 0E 01 BE 55 -> PC

AA 06 0e 06 01 c5 55 -> PC continuos read
aa 05 0e 00 BD 55 -> PC STOP
        

DISCONNECT #

aa 06 0e 06 06 00 c4 55 -> Risposta a STOP

SET TIME
AA 0a 01 17 09 1c 12 1d 33 53 55
10 1  23(anno) 9 (mese) 28 (giorno) 18 (ora) 29 (minuti) 

aa 04 06 b4 55
aa 05 0e 01 be 55

aa 05 0e 01 be 55 aa 05 0e 01 be 55

AA 05 0E 01 BE 55

brookst, 2024-04-23:
Following the notes from @Red-Owl, I managed to reverse most of the protocol to talk to this thing. I've put it into a single Python script as a Gist here: https://gist.github.com/brookst/bdbede3a8d40eb8940a5b53e7ca1f6ce
I'm still not understanding some of the log formats and there's a few commands I haven't explored. Most functionality is there and it can produce a once-a-second record of counts and doses etc.

@wnasich may be interested.

oOblik, 2024-05-02:
I'm also interested in FS-5000 support and willing to help out in any way I can.

Captainslarty, 2024-05-05:
Plus 1 from me - I would love to have rad pro on this unit.

oOblik, 2024-05-07:
Here are some additional PCB photos:
<details><summary>Photos</summary>
<p>

Front: 

[image]
Back:

[image]
Display:

[image]

</p>
</details>

Montecri, 2024-05-07:
Mine also came with this tiny tube, counts were like 1/3 of what you would get with a larger one. So, I replaced mine with an M4011 and, out of pure guesswork, removed the solder jumper R102. I guessed that it is used to inform which kind of tube is installed.

Put it all back together and counts from BS are within expected levels now.

Can anybody, who got one with the larger tube already installed, confirm if this jumper is opened or closed?

Thanks!

ihrapsa, 2024-05-07:
The smaller tube has both electrodes on the same side so what that jumper seems to do is bring the negative pole next to the positive one. Removing the jumper isn’t necessary when switching to the M4011 form factor tube.

Montecri, 2024-05-07:
> The smaller tube has both electrodes on the same side so what that jumper seems to do is bring the negative pole next to the positive one. Removing the jumper isn’t necessary when switching to the M4011 form factor tube.

Would you know how it detects it's a different tube and applies the correct conversion to uSv/h?

Tried to find a config for that and couldn't. It simply worked as expected after tube replacement, not only giving a much-improved CPM but also displaying a "correct" approximation of uSv/h.

Thanks!

brookst, 2024-05-07:
@Montecri 
> Can anybody, who got one with the larger tube already installed, confirm if this jumper is opened or closed?

Can confirm, R102 is open.
[image]

I'll take any excuse to take it apart.

ihrapsa, 2024-05-07:
> Would you know how it detects it's a different tube and applies the correct conversion to uSv/h?
>
> Tried to find a config for that and couldn't. It simply worked as expected after tube replacement, [...]

That is indeed weird. How are you checking wether the conversion/doserate is correct?

Montecri, 2024-05-07:
Took notes of BS (no real source here, unfortunately) before and after the conversion (CPM and uSv/h). Also, compared with other units I have (a GC01- running RadPro -  and a GQ 320+ v4).

Before conversion it had a much lower CPM but comparable uSv/h to other equipment, now it has both CPM and uSv/h comparable with the other ones.

Perhaps it measures some kind of voltage drop on the tube and can infer which type it is?

brookst, 2024-05-07:
> Perhaps it measures some kind of voltage drop on the tube and can infer which type it is?

I think it has to be something like that. I can see the trace from the R102 pads and all it does is go between the negative holder for the large tube and the through hole for connection to the negative lead of the smaller tube. I guess it might have needed a resistor there to reliably detect the smaller tube but didn't need it in the end.

Montecri, 2024-05-07:
> @Montecri
> 
> > Can anybody, who got one with the larger tube already installed, confirm if this jumper is opened or closed?
> 
> Can confirm, R102 is open. ![IMG_20240507_222609646_HDR](https://private-user-images.githubusercontent.com/659579/328680769-7df2aa25-8f18-4358-a740-8df4982bc360.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUxMTg0MjIsIm5iZiI6MTcxNTExODEyMiwicGF0aCI6Ii82NTk1NzkvMzI4NjgwNzY5LTdkZjJhYTI1LThmMTgtNDM1OC1hNzQwLThkZjQ5ODJiYzM2MC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwN1QyMTQyMDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZGRkNDA2MDEwZWMxMTRmMmUwODVkMTFkM2Y3NTVkYWY2ODdlYjM5ODU3NDk3ZmNkYmFmNTg4NzI0MmNmMjNkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.opM8MLUp8kCV4QnfEHXuZPcLkUh-4qg51EBFGwBdDtE)
> 
> I'll take any excuse to take it apart.

Thank you!

Perhaps this track (past the jumper) is also connecting to one of the pins on the MCU, which is pulled up internally, and it is used to determine tube type since I found no way on the UI to specify that.

Captainslarty, 2024-05-08:
I would be very surprised if it had ANY form of tube detection. The sweet spot range for virtually ALL tubes of this type is so similar that any particular setting is 'ok' - not optimised - but 'ok' ... so the device is pretty dumb as to the type of tube.

Captainslarty, 2024-05-09:
Gissio sir, if you do a custom firmware for this nice unit, then please unclutter the touch screen as it is horrendous. ;-) . too cluttered with too many icons that mean nothing really. Most could be controlled from the 3 button interface without the horrible touch screen, but careful use of the screen may be appropriate. It is a nice unit but the interface is pretty terrible and your marine interface design experience can add a tremendous amount.
Joe

brookst, 2024-05-09:
> Perhaps this track (past the jumper) is also connecting to one of the pins on the MCU, which is pulled up internally, and it is used to determine tube type since I found no way on the UI to specify that.

I don't see anything directly connected to the negative side of the tube(s). They go to ground through R59 and R62 on the front of the PCB. Q11 is connected to their midpoint, so I guess that is some kind of current feedback to the MCU. That is common to both tube types. The positive sides of both tube types are also directly connected to the HV supply. There's a chain of resistors to TP2 but that doesn't connect anywhere. There are traces from a few points in the HV supply, I don't know if they're just to control the tube voltage or if they're also used to measure the tube characteristics.

Captainslarty, 2024-05-10:
See above !!!! the processor ONLY has settings for the tube in it's program.  THERE IS NO WAY IT DETECTS A TUBE. Sheesh. ! ;-)

brookst, 2024-05-10:
@Captainslarty So how does it get the correct dose rate with different count rates? I believe that's what @Montecri is saying: The smaller tube has a lower count rate but still reports about the same dose rate.

How else could it get the conversion correct?

Captainslarty, 2024-05-11:
Hi Brookst, easy really. The manufacturer is aware what tube is placed in on the board when it is built so sets the PWM and Pulse width (mark to space ratio) in the code. It is the processor that provides the PWM for the voltage multiplier chain. That's it.
In the code written by Gissio you can control the frequency of the pwm and also the pulse width to provide more finely tuned options for different tubes although as I said above, they are all so similar that it you could replace the tube without any other mods if needed. If you were going to try to measure the voltage of the tube then you would need a VERY high impedance probe - circa 1 GIGA ohm !.  Using such a probe you often get a know fraction of the actual voltage as the result.
as most even good quality multimeters have an input impedance of 100th of that the reading it with a normal meter would be way off as it would load the circuit down. Often manufacturers put a chain of high value resistors to a test point - say 50 mega ohm or similar, that forms a voltage divider with the usual 10 mega ohm termination for the tube, this gives a rough and ready value they can test but it is nowhere near the value actually at the tube, just a voltage that they know would very coarsely  correspond to the correct value if used with a suitable measuring probe. It is all set in the firmware at the time of building.
Joe

Captainslarty, 2024-05-11:
Just to add to the discussion. I don't think that changing the tube altered anything. These tubes are - as said, much of a likeness in requirements. The smaller tube comes from different manufacturers and there are reasonable and downright awful types. ;-) . The count rate is 'near enough' at low levels. Comparing to the gc-01 is not really a great idea as in it's basic form it is a bit of a lottery where, in that case, the smaller tube is absolutely 'deaf'. (results compared to my radiacode 102). My Bosean 5000 with the large tube from the factory is the nearest to the radiacode. the GC-01 with the small tube is a bit of a joke. I have a 4011 to go in it and set it up with rad pro and with an extremely high impendence probe for the HV and it will be very adequate I am sure. The Bosean from the factory with the J321 tube in will almost certainly have a few parameter differences in firmware which you cannot, at the moment, without rad pro - change.
Joe

Captainslarty, 2024-05-11:
Here is a good note for measuring tube voltages with a normal 10M ohm input high quality multimeter. (basically without a lot of other options - you cannot). Here is how to produce a series resistor of known value of around 1 GIGA ohm to use with a quality known input meter.
https://www.giangrandi.org/electronics/gigaohm/gigaohm.shtml

Montecri, 2024-05-11:
Just to add data to the topic When I received the Bosean FS-5000, my first impulse was NOT to crack it open, since I was already having fun with GC-01. However, I noticed immediately that the CPM was way down compared to GC-01 and GQ 320+, both using M4011 tubes.

Like, the three turned on, side by side, for some time, the CPM between GC-01 (running RadPro) and GQ 320+ for background source floated between 24 and 33 where I live; while for the Bosean FS-5000 it was between 8 and 11, consistently 1/3 of what I could observe on the other two, while the uSv/h remained more or less the same among them.

That's what prompted me to open the Bosean and see what was going on and, to my disappointment, found the tiny tini tube no one wants. Had a spare M4011 and immediately replaced it, removing the R102 jumper first, since intuition told me that (ha!); now we know it doesn't play any role in tube selection.

Now, looking at the official Bosean store listing for the same item I see this:

"We Promise: We will replace all products in the warehouse and produce new items with NEW J321 tube to ensure all items are J321. April 10, 2024."

Thanks!

Captainslarty, 2024-05-12:
Yes, that is why I bought mine - only after the promise of the J321 made me buy one. I am glad I did as it is a great unit. I don't like the touch screen but hey ;-). The 321, 305 and 4011 tubes are all basically identical specifications. You made a wise move replacing the tube. My GC-01 from Banggood came with the smaller tube and the advertising clearly showed a larger - full size - tube. I obtained a refund from them from the price of the tube from aliexpress (18.29 euro). Measuring a known source like radium is a great test, or even the small amount of gamma from thorium (Ex russian military compass for the radium and a lantern mantle and coated welding rods containing thorium for the second). Joe

koogar, 2024-05-16:
> See above !!!! the processor ONLY has settings for the tube in it's program. THERE IS NO WAY IT DETECTS A TUBE. Sheesh. ! ;-)

Indeed the component values for the high voltage section look the same for both the J321 and the smaller tube (J612?) 
Apart from R102 which is a jumper for the smaller tube. 

There seems to be some software configuration data for the specific tubes, as the J321 shows "Sensor 3" in the device information and the smaller one "Sensor 1" , there may even be a "Sensor 2" configuration.

Gissio, 2024-05-16:
> Here are some additional PCB photos:
> 
> Photos

Can I use your photos in Rad Pro?

koogar, 2024-05-16:
> > Here are some additional PCB photos:
> > Photos
> 
> Can I use your photos in Rad Pro?

If you need any photos I have many of the j321 and j612  Bosean FS 5000's you can use

oOblik, 2024-05-17:
> > Here are some additional PCB photos:
> > Photos
> 
> Can I use your photos in Rad Pro?

Of course, anything you need. I'm glad to help out in any way I can.

Gissio, 2024-05-18:
> > > Here are some additional PCB photos:
> > > Photos
> > 
> > 
> > Can I use your photos in Rad Pro?
> 
> If you need any photos I have many of the j321 and j613 Bosean FS 5000's you can use

Sure, one for each tube, with the case open would be great!

koogar, 2024-05-20:
> > > > Here are some additional PCB photos:
> > > > Photos
> > > 
> > > 
> > > Can I use your photos in Rad Pro?
> > 
> > 
> > If you need any photos I have many of the j321 and j613 Bosean FS 5000's you can use
> 
> Indeed, one for each tube, with the case open would be great!

Sure here you go https://github.com/koogar/Bosean-FS-5000-Radiation-Detector

Montecri, 2024-05-20:
> Sure here you go https://github.com/koogar/Bosean-FS-5000-Radiation-Detector

Seeing your pictures it shows for J613 as "Sensor 001" and J321 as "Sensor 003". I changed mine J613 for an M4011 (J321) and removed jumper R102. Just leaving this information here so future readers understand that the tube type is hardcoded on the firmware, apparently. No matter if you change the tube or tweak R102.

Regards,

- Cristiano

koogar, 2024-05-20:
> Seeing your pictures it shows for J613 as "Sensor 001" and J321 as "Sensor 003". I changed my factory-installed J613 for an M4011 (J321), removed jumper R102 and it is still shown as "Sensor 001". Just leaving this information here so future readers understand that the tube type is hardcoded on the firmware, apparently. No matter if you change the tube or tweak R102.
> 
> Regards,
> 
> * Cristiano

Yes I meantioned that further up the thread :)

A M4011 is similar to a J321 so maybe that is a "Series 2"

Regards

Rupert

ricsim, 2024-05-26:
I just received mine with a J321 tube.
I have seen that 2.0rc5 supports it, and that there is a device section for it.
But the link still goes to 2.0rc4. Any link where I can get the FS5000 SW before rc5 is released?
MANY thanx for this! I also use it with a Fnirsi GC-1 which made it usable, as the HV was set waaaay to high, which produced too high counts. With RadPro I could fix the HV settings!

Gissio, 2024-05-26:
Rad Pro is already working on the Bosean FS-5000. Hang on, it's just a couple of days for 2.0rc5!

ihrapsa, 2024-05-26:
> I just received mine with a J321 tube.
>
> I have seen that 2.0rc5 supports it, and that there is a device section for it.
>
> But the link still goes to 2.0rc4. Any link where I can get the FS5000 SW [...]

Latest PR added github actions support. Checkout under [Artefacts](https://github.com/Gissio/radpro/actions/runs/9240740459) of the latest build.

Gissio, 2024-05-28:
2.0rc5 with Bosean-FS5000 support is out!

Closing this issue.

closed 2024-05-28 by Gissio

Gissio, 2025-05-21:
Iona: it my understanding that you are a native Romanian speaker. Could you check the Romanian translation in 2.2beta3?

The strings are available here: https://github.com/Gissio/radpro/blob/main/platform.io/src/strings/ro.h

If you are willing to help us with these, please follow the discussion here: https://github.com/Gissio/radpro/discussions/241

ihrapsa, 2025-05-21:
Hey, I'll be more than happy to help!

Gissio, 2025-05-22:
Thanks for the PR! Can you respond in #241 ?