Gissio/radpro #12: HFS-P3 support

ihrapsa, 2023-11-18:
[image]
This is a nice little pen-like device that apparently could benefit from a lot of firmware improvements:
- persistent cumulative dose after shut down
- cps/cpm monitoring
- pulse click with ability to enable/disable this
- high dose-rates measuring improvments
- temperature monitoring
- power saving features
- etc.

[Here](https://radmon.org/index.php/forum/commericial-geiger-counters/1287-hfs-p3-pen-geiger-counter-dosimeter) is a nice an thorough review of it.

**MCU:** [ARM Cortex-M0 FM33LC043N MCU from Fudan Micro](https://www.lcsc.com/product-detail/Microcontroller-Units-MCUs-MPUs-SOCs_FUDAN-span-style-background-color-ff0-MICRO-span-FM33LC043N_C908397.html)
**Sensor:** HH614 GM Tube
**Battery:** 150mAh LiPo cell - 50h promised

<details>
<summary>Click for internal pics: </summary>

PCB layout [from radmon.org forum]

[image]  

SWD pins layout under the battery tape [from radmon.org forum]

[image]

</details

I'm urrently waiting for a SWD debugger to play with this. Let me know what else I can provide and how I should go about this.

closed 2024-05-09 by Gissio

ihrapsa, 2024-06-08:
Sorry for reviving this but I managed to dump the firmware of this device. SWD port is not locked as previous thought.
I used a J-Link device (actually a `ST-Link V2` clone flashed with `J-Link` using `STLinkReflash`) and I managed to dump it with `J-Link Commander` as well as with `Keil µVision V5` 

For **J-Link Commander** I dumped it with   
```savebin C:\Users\jonah\Downloads\HFS-P3_dump.bin, 0x0, 40000```
<details>
  <summary>J-Link Commander Screenshot</summary>

[image]

</details>

For **Keil µVision V5** I dumped it with 
```SAVE HFS-P3_dump.hex 0x0, 0x3FFFF``` Then converted the Intel HEX-386 file format to binary using [hex2bin.exe](https://hex2bin.sourceforge.net)
<details>
  <summary>J-Link Commander Screenshot</summary>

[image]

</details>

[HFS-P3_dump.bin.zip](https://github.com/user-attachments/files/15748349/HFS-P3_dump.bin.zip)

Gissio, 2024-06-08:
The FM33 is quite different from the STM32. This would require a lot of time and motivation, which I don't have.

ihrapsa, 2024-06-08:
I can understand that. Thanks anyway! 🙂 
In the meantime I found some interesting developing sources for FM33LC0xx here on GitHub. In case anyone else is brave and knowledgeable enough to hop on this I'm willing to help with testing.

https://github.com/xdu-aero-association/FM33LC0XX_Example_0.04/tree/master
https://github.com/SoCXin/FM33LC0

PS. I was thinking about posting all this in that radmon.org [post](https://radmon.org/index.php/forum/commericial-geiger-counters/1287-hfs-p3-pen-geiger-counter-dosimeter#6729) as well but I'm having issues creating an account.

Gissio, 2024-06-11:
One more thing... could you check #97 ?

ihrapsa, 2024-06-11:
> One more thing... could you test #97?

Will sure do!

ihrapsa, 2024-08-11:
#### [UPDATE] 
I manged to trace the board and create a schematic. 🥳 
Thanks to a lot of resources online I manged to build a base project (**Keil uVision IDE**) and control the OLED (over bit-banged I2C) and LEDs. All other GPIOs are mapped and can be controlled as well.

Not sure how to go about sensing the tube counts and all the extras (menu creation, calibration, battery management, internal temp sensor etc.).

I made a GitHub repo with the current state of the IDE project and all the resources I used.
https://github.com/ihrapsa/HFS-P3_DevKit

Any help would be really appreciated!

kenoro, 2024-11-01:
Really nice work, don't get influenced by the entitled people on radmon. Just got mine now.

ihrapsa, 2024-11-04:
Haha, yeah, thanks! What happened there was unexpected but I got the point 😅

Project got halted for a bit due to life, but there is still stuff to figure and build (Menu UI, build the logic behind counting and averaging counts, conversion to doserate, alarms etc)

Cyl0nius, 2025-04-03:
Any news about custom firmware for this handy pen geiger counter?