Gissio/radpro #13: YT-203B Temperature

mulcmu, 2023-11-22:
I flashed [2.0beta11](https://github.com/Gissio/radpro/releases/tag/2.0beta11) onto two YT-203B with GD32F150 MCUs.  The device temperature shown on the statistics screen for both units fluctuates between 320 to 323 °C at room temperate (about 19°C).

Gissio, 2024-01-12:
That seems like a hot issue! Will be fixed in the following beta. Please watch the project so you get notified when there are news.

Gissio, 2024-02-17:
I have just released 2.0beta14. Could you check if it works now?

mulcmu, 2024-02-24:
I just tried beta16 on one of these.  Better but still running a bit high.  Reading is around 30.0°C or so and pretty noisy.  Ambient was about 19°C again.

Gissio, 2024-02-24:
Thanks for the report. I'm glad it works now. Remember, the temperature is device temperature (measured at the MCU), not ambient temperature, so 30.0°C seems allright!

I'll keep this issue open as I'm still planning some changes in the coming days.

Gissio, 2024-02-25:
I forgot to mention. There are new Contrast and Brightness settings. Do they work? Did contrast improve the display?

mulcmu, 2024-02-25:
Contrast and brightness both worked well.  Setting 3 on the contrast was perfect for test unit.  Level 8 was solid black which would make it difficult to adjust if it was errantly set that high and menu was exited.  

There might be a regression with the battery indicator.  It was showing empty on beta16, cells measured 2.85 volts on DMM while installed.

Gissio, 2024-03-10:
I have just released [2.0beta17](https://github.com/Gissio/radpro/releases). Can you see if the battery indicator works now?

Gissio, 2024-03-14:
Repeat of https://github.com/Gissio/radpro/issues/21.

Should be fixed in beta19.

closed 2024-03-14 by Gissio

mulcmu, 2024-03-14:
[2.0beta19](https://github.com/Gissio/radpro/releases/tag/2.0beta19) tested and working great

Gissio, 2024-03-14:
I'm glad it works now!