Gissio/radpro #14: Is it possible to connect this to PC and collect data?`

NikoTumi, 2024-01-13:
I want to run a script on my PC that would email and SMS me data log of when the measurement goes +X%.

Is it possible to grab the readings or logs from the GC-01. 
This is the one I got.
https://www.amazon.com/Geiger-Counter-Nuclear-Radiation-Detector/dp/B0BHH9X1WG/ref=sr_1_4?keywords=nuclear+radiation+detector&sr=8-4&ufe=app_do%3Aamzn1.fos.18ed3cb5-28d5-4975-8bc7-93deae8f9840

Thanks for the help.

Gissio, 2024-01-13:
FNIRSI GC-01 support is not ready yet (but it won't take long).

Once it is, yes, you will be able to log data.

Gissio, 2024-01-13:
FNIRSI GC-01 support is not ready yet (but it won't take long).

Once it is, yes, you will be able to log data.

In the meantime, watch the project to get notified when a new release comes out.

NikoTumi, 2024-01-13:
So quick research on the chip being used, CH32F103 R8T6, and found the following.
Chip: CH32F103 R8T6
Existing Repo: https://github.com/IOsetting/StdPeriphLib_CH32F1
Detailed manual: https://crossic.com/product/ch32f103r8t6/
I'll look more into this.

Gissio, 2024-01-13:
Don't worry, Rad Pro for GC-01 is really around the corner!

JantarX, 2024-02-19:
hi.  first of all, I want to thank you for the work on those devices.  but there is a boot error on the fnirsi GC-01 device with firmware rad pro 2.0 beta.  the device does not boot the firmware, but fortunately Windows can still open it and upload the old firmware to it.  I had firmware 1.6 since new, now I have 1.6-2 which I downloaded from the net but it looks like it multiplies the dose.  Do you happen to have the original 1.6 installed somewhere?  well thank you

Gissio, 2024-02-19:
Unfortunately I only have firmware 0.3 and 1.5, which you can find here: https://github.com/Gissio/radpro/blob/main/docs/install-gc01.md.

Can you refer me to a link to firmware link 1.6?

JantarX, 2024-02-20:
Ofcourse. Here is it: [](https://drive.google.com/drive/folders/1XN-d1ZiFq0ur_ae392iQr-pbHZY32Ag5)

But I don't have the clean version 1.6. That's what I'm looking for. There is only 1.6-2 where there is an extended translation and then in 1.5 where fonts are missing after reboot unless I activate them in the settings by selecting the language. The problem is that in version 1.6-2, my background dose jumped double. I don't know if it's real, I have nothing to compare it with. It would be ideal if someone who has two GM computers tested it, and even more ideally if they compared it to one that is professionally calibrated.

Gissio, 2024-02-22:
I have added a new release. Can you try now?

JantarX, 2024-02-22:
Ofcourse. I want to know how the measurement will go with your firmware :)

Gissio, 2024-02-22:
Thanks for your patience!

I'm sending you a special test firmware:

[radpro-beta17.test1.bin.zip](https://github.com/Gissio/radpro/files/14377954/radpro-beta17.test1.bin.zip)

After flashing it, unplug from USB.

Keep the ok/power key pressed. The pulse LED should flash 13 times. If it flashes fewer times, please let me know how often it flashes. This will let me diagnose the problem. 

Now you can release the ok/power key. You should see a screen similar to this one:

[image]

If you get to this point, please send me a photo of the screen.

mizsha, 2024-02-22:
Hi, same trouble, my device flashes 13 times, then restarts ... with FW in zip file.

Gissio, 2024-02-22:
That's actually great news: all the peripherals are being initialized correctly.

What do you mean with "then restarts ..."? Nothing else happens? Or does the 13-flash sequence start again?

Could you try releasing the power key after the fourth flash and see what happens (the 13-flash sequence should continue).

Gissio, 2024-02-22:
That's actually great news: all the peripherals are initializing correctly!

I've just found a very silly error with the LCD. Again, thanks for the patience:

[radpro-beta17.test3.bin.zip](https://github.com/Gissio/radpro/files/14379748/radpro-beta17.test3.bin.zip)

Even if the screen does not work, let me know whether:

* The LCD backlight turns on (it should be fairly dim)
* The pulse LED flashes periodically.

mizsha, 2024-02-23:
[image]
LCD flashes with some artefacts and LED dont turn on or even pulse.

Gissio, 2024-02-23:
It seems the LCD display is receiving commands now, but something is still off.

Let's do one more test: as before, the output should be the white screen with four lines I showed you on my previous message. The LED light and the buzzer should pulse periodically.

[radpro-beta17.test4.bin.zip](https://github.com/Gissio/radpro/files/14381534/radpro-beta17.test4.bin.zip)

Again, thanks for your patience.

JantarX, 2024-02-23:
radpro-beta17.test4 - now there is only a white screen, LED and buzzer that blink irregularly. No characters, no image failure, just a lit panel.... That's okay. I guess we can fine-tune it together somehow :) ... It looks like me and Mizsha have the same device

JantarX, 2024-02-23:
radpro-beta17.test4 - now there is only a white screen, LED and buzzer that blink irregularly. No characters, no image failure, just a lit panel. 
It looks like me and Mizsha have the same device because the previous test turned out the same. 

That's okay. I guess we can fine-tune it together somehow :)

Gissio, 2024-02-23:
Let's try one more time. This time I'm sending two tests. Let me know which one(s) work.

[radpro-beta17.test5.and.test6.zip](https://github.com/Gissio/radpro/files/14389293/radpro-beta17.test5.and.test6.zip)

Gissio, 2024-02-23:
Can you make a short video?

JantarX, 2024-02-23:
https://github.com/Gissio/radpro/assets/140242284/e3c66e64-85ea-4ccf-9b1d-3b4917b54347

After an even longer pause, it loads for 15 seconds.

JantarX, 2024-02-23:
[image]
[image]

I'll attach photos of the board if you happen to need to compare them

Gissio, 2024-02-23:
That was fast! Thanks for the video and photos.

It seems the device is resetting at the moment the screen turns off. I'll get in touch once I have news.

JantarX, 2024-02-23:
I just put it together when you asked for a video so it went quickly. Great, I’m keeping my fingers crossed for you.

Gissio, 2024-02-23:
I think I found the culprit! In case you are interested: the code that reads the ADC never exited, so the watchdog timer kept restarting the device before writing to the screen.

Let's try once again:

[radpro-beta17.test7.bin.zip](https://github.com/Gissio/radpro/files/14390369/radpro-beta17.test7.bin.zip)

mizsha, 2024-02-24:
[JantarX](https://github.com/JantarX) They look similar, I also have a version with J321 tube, later versions come with cheaper one, but it also looks like we have a different LCD module and I don't even have the option and pins on the board to connect a smaller and cheaper tube. I have one of the first versions sold by a local dealer, not the current one from ali. I am attaching a photo.

[image]

Does anyone know how to successfully upload new firmware in linux ? BOOT partition is visible, readme.txt is visible, fw can be uploaded, but initialization and restart is not done, after disconnection the old fw is still active. What is the difference between uploading in win and linux, does windows have any special device driver ? 

I don't have a chance to reboot into win for another test today :(

mizsha, 2024-02-24:
Photo of one of the latest versions from aliexpress. The same box, different hw, it seems like fun for debugging fw, wouldn't it be useful to use some debug port to get data ? Trial and error will take ages.
[image]

JantarX, 2024-02-24:
> I think I found the culprit! In case you are interested: the code that reads the ADC never exits, so the watchdog timer kept restarting the device before writing to the screen.
> 
> Let's try once again:
> 
> [radpro-beta17.test7.bin.zip](https://github.com/Gissio/radpro/files/14390369/radpro-beta17.test7.bin.zip)
> 
> If it works, please press the keys (one at a time) and see if GPIOA and GPIOC change.

It is restarting without any changes compared to tests 5 and 6.

JantarX, 2024-02-24:
> [JantarX](https://github.com/JantarX) They look similar, I also have a version with J321 tube, later versions come with cheaper one, but it also looks like we have a different LCD module and I don't [...]

Have you tried version 1.6-2 that I shared above? Does it work on both of your devices?

As for Linux, I personally can’t help you. The only thing I know is that you literally need Windows 10 and a USB 2.0 port to run the bootloader. I tried it on USB 3.0 but it didn’t launch. I tried it on Ubuntu and the whole bootloader crashed. I had to disconnect the battery and flash it according to the conditions above.

mizsha, 2024-02-24:
I tried 1.6-2 and all subsequent test versions only on a device with J321
tube, with the same result as you have. I don't have the other device
available, I only have a photo.

More info for linux ( I have no experience with STM32 devices, maybe it is
trivial, it would help me in further testing. ):

$ lsusb
Bus 001 Device 026: ID 0483:002a STMicroelectronics AIR01

$ fdisk
Disk /dev/sdc: 100 MiB, 104857600 bytes, 51200 sectors
Disk model: DETECTION 1.0
Units: sectors of 1 * 2048 = 2048 bytes
Sector size (logical/physical): 2048 bytes / 2048 bytes
I/O size (minimum/optimal): 2048 bytes / 2048 bytes

105 MB — 82 KB free
FAT (12-bit version)

stevemisel, 2024-02-24:
Hey folks … have been watching this thread for a week or two .. am
available to do some testing….

Gissio, 2024-02-24:
I'm still unable to reproduce the same problem locally.

One more test. Please make a video until you hear nothing (it should take over a minute to finish). The beeps encode the CPU's state at the fault.

[radpro-beta17.test8.bin.zip](https://github.com/Gissio/radpro/files/14394146/radpro-beta17.test8.bin.zip)

stevemisel, 2024-02-24:
> I'm still unable to reproduce the problem locally.

I know there was, at one time, some documentation on flashing the GC-01 ...
but it appears to be gone.   If you can point me to it, I'll give it a
whirl...

Gissio, 2024-02-24:
stevemisel: for now let's try fixing the problem with the bootloader.

Splinter34, 2024-02-25:
Been following this project...  Great work here!  This is the latest test bin.  Unit board marked  V.03.  It is recovered from error with V1.6-2 easily after pressing the reset button.  The ticks indicate a led flash, beeps are beeps.  Looking forward to the finished project.  Thanks Gissio!  If you need anything else just let me know.  I can also jtag this board if needed to get the memory.

https://drive.google.com/file/d/1X46jFWtaB2KieWibZ7GwSeiXOtlxqoTg/view?usp=sharing

Gissio, 2024-02-25:
Thanks for the kudos, always appreciated.

So it seems USB is causing the troubles. I turned it off now. Let's hope this is it.

[radpro-beta17.test9.bin.zip](https://github.com/Gissio/radpro/files/14397452/radpro-beta17.test9.bin.zip)

Splinter34, 2024-02-25:
Maybe closer ?  Has a white screen now. Before no screen.  Audio slightly muffled, If you need me to try to get better video let me know.

https://drive.google.com/file/d/1InQxWHiGRQz2QJMyTfZSuyTbtAIiYCFc/view?usp=sharing

Splinter34, 2024-02-25:
By the way that is flashing on u USB 3.0 port.  Do I need to find something USB 2.0?  Thought I read something somewhere about that?

Gissio, 2024-02-25:
That worked! It's a buffer overflow. I'll keep you updated.

Gissio, 2024-02-25:
In the meantime: with test9, could somebody make a video of the tones before the screen turns on? Please place your microphone as close as possible to the device so I can time things.

Splinter34, 2024-02-25:
I'll do it right now.

Splinter34, 2024-02-25:
There is no tones before the screen lights up.  If you want I can unsolder a wire so there will be no clicks.  Let me know.
Test 9 ver2 video:
https://drive.google.com/file/d/1CenxAx_cgtsKUn22eVM4kAB1HekQ0LoW/view?usp=sharing

Gissio, 2024-02-25:
My bad! Can you try again?

[radpro-beta17.test10.bin.zip](https://github.com/Gissio/radpro/files/14397770/radpro-beta17.test10.bin.zip)

Splinter34, 2024-02-25:
https://drive.google.com/file/d/1fEcoXsikp3MEqo2Adp2z2rF-KL8VBbVe/view?usp=sharing

This had tones before the screen lit.....

Tig3rch3n, 2024-02-27:
[radpro-beta17.test10.bin Audio.zip](https://github.com/Gissio/radpro/files/14418330/radpro-beta17.test10.bin.Audio.zip)

One Suggestion, because the small Tube only has 1\2 to 1\3 of the sensitivity of the j321 tube, my guess is they adjusted that with the 1.6-2 Firmware Version.

I have a Tile with Uraniumglasing
Both devices in the same Orientation, on the same spot did differ A LOT.
Small Tube max 1.51µS/h
J321 Tube max 9.77µS/h

I have both, the small cheap Tube and the J321 Tube
Both PCBs look the same and AFAIK they behave Identical with the Test files.

Splinter34, 2024-02-28:
Here you are...
[GeehyDump.txt](https://github.com/Gissio/radpro/files/14426970/GeehyDump.txt)

Tig3rch3n, 2024-02-28:
Jup... They differ...
[Memory Dump.txt](https://github.com/Gissio/radpro/files/14430313/Memory.Dump.txt)

08003800: eb 09 07 00 01 01 01 00 00 00 00 00 00 00 00 00
08003810: 00 00 00 00 40 01 00 00 64 00 d0 07 e6 07 01 01
08003820: dd 0f 00 00 98 b7 f4 01 ff ff ff ff ff ff ff ff

versus

08003800: eb 04 07 00 01 01 01 00 00 00 00 00 00 00 00 00
08003810: 00 00 00 00 37 23 00 00 c8 00 d0 07 e8 07 02 04
08003820: 76 22 02 00 98 b7 f4 01 ff ff ff ff ff ff ff ff

Gissio, 2024-02-28:
Try resetting the device with the reset button. Otherwise the firmware will drain your battery.

JantarX, 2024-02-29:
> BTW, I just learned that the V1.6-2 firmware only works on GC-01's with Geehy processors.
> 
> WCH processors have 64 kB flash; Geehy ones, 128 kB. So, quite simply, V1.6-2 doesn't fit in the flash of WCH processors.

Mizsha tested in 1.6-2 and also on the version with the WCH processor, and it worked for him. He was only interested in the J321 tube.

By the way, I apologize for disappearing for a while. I was dealing with drivers that work with Windows 11.

JantarX, 2024-02-29:
> Jup... They differ... [Memory Dump.txt](https://github.com/Gissio/radpro/files/14430313/Memory.Dump.txt)
> 
> 08003800: eb 09 07 00 01 01 01 00 00 00 00 00 00 00 00 00 08003810: 00 00 00 00 40 01 00 00 64 00 d0 07 e6 07 01 01 08003820: dd 0f 00 00 98 b7 f4 01 ff ff ff ff ff ff ff ff
> 
> versus
> 
> 08003800: eb 04 07 00 01 01 01 00 00 00 00 00 00 00 00 00 08003810: 00 00 00 00 37 23 00 00 c8 00 d0 07 e8 07 02 04 08003820: 76 22 02 00 98 b7 f4 01 ff ff ff ff ff ff ff ff
> 
> [EDIT] Stupid question... How do I get back into Storage Mode do flash back the 1.6-2 Firmware?

I’m dealing with a bootloader failure by disconnecting the battery and starting the bootloader without the battery. So far, it has always worked.... Specifically, the battery for the main power supply, not the small one on the board.

JantarX, 2024-02-29:
> I have news.
> 
> First, I found no buffer overflow in the code.
> 
> Second, things are working on GC-01s with a WCH processor (CH32F103R8T6) but, apparently, not with a Geehy one (APM32F103RBT6). Fortunately I happen to have a board with a very similar Geehy processor (APM32F103CBT6), so I repurposed it and, alas, Rad Pro worked there. So, being the environments so similar, the only thing that could make a difference is the bootloader. So I made this special firmware that should let you dump the bootloader:
> 
> [radpro-2.0beta17-test12-bootloader-dump.bin.zip](https://github.com/Gissio/radpro/files/14426035/radpro-2.0beta17-test12-bootloader-dump.bin.zip)
> 
> After installing and re-starting your GC-01, a USB serial port device should appear on your computer (you might need to unplug/replug the device). Under Windows, open the Device Manager, go to the "Ports (COM & LPT) section and take note of the COM port. Next, download PuTTY (https://the.earth.li/~sgtatham/putty/latest/w64/putty.exe), start it, click on "Serial" and type the COM port in "Serial line". A terminal window should show up. Press ENTER and the memory dump should start. Next, go to the window menu (upper left), select "Copy All to Clipboard" and paste in a .txt file. Attach the file to a message in this thread.
> 
> Even if you have a GC-01 with a WCH processor, or somebody else already uploaded the dump, it would be interesting that you repeated the process, just in case there are different versions of the bootloader.

The memory test of the bootloader is not working for me at all. The device dies with the LED on, but the bootloader does not work. I tried to connect it even without invoking the bootloader with hardware buttons, but to no avail. The COM port is empty, PUTTY is just waiting.

Gissio, 2024-03-10:
I have news.

In short: go and grab the [latest release](https://github.com/Gissio/radpro/releases)!

In long: the GC-01 bootloader is awful. On STM32, bootloaders are supposed to provide a correct environment. The CH32/APM32 (Geehy) processors have 20 kB RAM, and the stack pointer should be pointing at the end of RAM. The GC-01 bootloader set up the stack pointer at the beginning, so, of course, stack overflows would occur! There is also a USB bug with CH32 processors: it turns out that the CH32's USB hardware does not wake up the processor, while it should; that's why USB was not working with CH32 processors.

JantarX, 2024-03-10:
Thank you for the new release, but GitHub servers for downloading files from repositories are currently blocked by my provider and browsers due to recent hacking attacks. Can you put them somewhere else

Gissio, 2024-03-10:
Sure: https://drive.google.com/file/d/1f8R15QnVgOWdbKa3C3Iw5216w2PU-889/view?usp=sharing

Arparela, 2024-03-10:
hi, I just flashed the latest version, but I have this result, I attach a photo of my motherboard
https://github.com/Gissio/radpro/assets/162920409/26dc17b4-49d6-4da4-a622-b5de6507e490

[image]

Gissio, 2024-03-10:
Arparela, welcome.

What is the result?

Tig3rch3n, 2024-03-10:
Same Problem as Arparela, white Screen with a few beeps and then a constant Noise

Gissio, 2024-03-11:
Identified and corrected these bugs:

* CRC calculation error (the beeps).
* Display not addressed correctly (the white screen).
* Pulses were not detected correctly (the noise).

Can you try again? Use the appropriate file for your device (CH32F103R8, APM32F103RB Geehy). Don't play with the HV settings for now.

[radpro-fnirsi-gc01-2.0beta17-test2.zip](https://github.com/Gissio/radpro/files/14553128/radpro-fnirsi-gc01-2.0beta17-test2.zip)

Splinter34, 2024-03-11:
It works!  great job Gissio!  Well except for that the screen display is upside down.  (minor issue)  Thats awesome!

Gissio, 2024-03-11:
This should work even better (fixes upside down display):

[radpro-fnirsi-gc01-2.0beta17-test3.zip](https://github.com/Gissio/radpro/files/14553430/radpro-fnirsi-gc01-2.0beta17-test3.zip)

Can you please let me know what microcontroller your GC-01 has (CH32 or Geehy) and perform the following tests?

- Battery indicator works when not charging.
- Battery indicator works when charging.
- Display contrast/brightness settings work.
- Alarm works (flashes the screen, beeps, triggers the vibrator).
- Device clock works.
- Random generator works (it takes two pulses to generate one bit of data).
- Dead-time measurement works (takes some hours to converge).
- USB data connection works (turn on device with USB not plugged in, plug in later, a serial port device should appear).
- radpro-tool works with the serial device.
- Data logging works (read datalog with radpro-tool).
- Temperature readout in the Statistics is correct.

Also, can you try the following?

- Go back to the original firmware. With a regular multimeter (500 V setting), measure the HV voltage at the tube (Yes, I know, the measurements is very bad, but I just want to do a relative measurement). Be very careful as high voltage can be LETHAL. Install Rad Pro. Go to "Geiger tube" menu, select "HV profile", select "Custom", set 10 kHz PWM frequency and change the PWM duty cycle until you get the same voltage with the original firmware. Please repeat the last steps with 2.5 kHz PWM frequency (no need to re-install original firmware).

w01fdev, 2024-03-11:
Hello

great project

radpro-fnirsi-gc01-2.0beta17-test2.zip was the first version that worked for me. with radpro-fnirsi-gc01-2.0beta17-test3.zip I notice:

Geehy APM32 - F103RBT6

- Display brightness doesn't work
- display sleep remains white bright screen

I haven't tested the rest yet.

regarding orders on aliexpress:
I recently ordered mine from aliexpress last month. By the way, the fnirsi store was very nice in communicating and sent me one with a J321 tube because I insisted on it and said that otherwise the device would be returned.

[image]
[image]
[image]

[29 more replies]

closed 2024-04-15 by Gissio