Gissio/radpro #17: Bosean FS600 2.0beta16 Dsiplay is black

Dsl71, 2024-02-25:
I dried the new 2.0 beta16 on a board revision KL7 820.403D. The display had all pixels set to black, no background light, text was only readable in a very steep angle. keyboard seems to fail also. Klicks were hearable, the LED was flashing also.

Gissio, 2024-02-25:
A regression, how sad. Does this work?

[radpro-bosean-fs600-2.0beta17-test1.bin.zip](https://github.com/Gissio/radpro/files/14397993/radpro-bosean-fs600-2.0beta17-test1.bin.zip)

Dsl71, 2024-02-25:
this is only 32k instead of 128k, is this okay to flash?

Gissio, 2024-02-25:
Yep. It'll complain with a checksum error because the firmware is not signed, but other than that, it should work.

Gissio, 2024-02-25:
Oh, the settings might be wrong, but you can correct them.

Dsl71, 2024-02-26:
Just flashed it. After powering, the led flashes about 5 times and then it starts with checksum error display and then it shows everything normal. Btw, first time to see an output from the random generators. Cudos to you, Gissio ;-)

Gissio, 2024-02-26:
I'm glad it works now.

Do the new display contrast/brightness settings work? Does the dead-time measurement (statistics) work? Is the battery indicator working?

Dsl71, 2024-02-29:
After about 11h runtime dead-time shows < 1.7 µs

Dsl71, 2024-03-02:
Bosean FS-600:
After about 46h runtime dead-time shows < 0.0 µs

Gissio, 2024-03-10:
I just released [2.0beta17](https://github.com/Gissio/radpro/releases). It should fix the dead-time measurement and battery level readout. Also, settings should work now.

Dsl71, 2024-03-10:
hmm, flashed it , but only the led is red all the time, no display text

Gissio, 2024-03-11:
Can you make a video and share it here?

Dsl71, 2024-03-11:
Uff, even a 5 sec video is too big. so:

- Press on
- Led lits constantly
- No screen light and text
- if  a click apears, short led off.

thats all

Dsl71, 2024-03-11:
https://github.com/Gissio/radpro/assets/57271038/a366396e-46e1-445b-8007-94acaee67f55

Gissio, 2024-03-11:
That video helped a lot, thanks! I'll keep you updated.

Gissio, 2024-03-13:
I just released Rad Pro 2.0beta18. Can you go over this checklist?

- The display/backlight works.
- Settings are stored to flash memory.
- The radiation alarms work (flashes the screen and beeps).
- While unplugged, the battery indicator works correctly.
- While plugged, the battery indicator shows a lighting icon.
- The temperature indicator in the statistics menu shows reasonable values.
- The random generator and dead-time compensation work.
- The serial port communication works.
- Data logging works and you can download it with radpro-tool.

Dsl71, 2024-03-13:
YES-  The display/backlight works.
    NO- Settings are stored to flash memory.
    The radiation alarms work (flash the screen and beep).
    YES- While unplugged, the battery indicator works correctly.
    YES- While plugged, the battery indicator shows a lighting icon.
    NO*-The temperature indicator in the statistics menu shows reasonable values.
    YES**- The random generator and dead-time compensation work.
    TBD- The serial port communication works.
    TBD- Data logging works and you can download the datalog with radpro-tool.

* Crash while entering menu,have battery to remove to reboot
** How can i check if dead-time compensation work

Gissio, 2024-03-14:
I'm sorry, I meant dead-time measurement, not dead-time compensation. If you can't enter the statistics screen, you can't do this for now.

Can you try dumping the contents of the flash memory to a file and upload it here? I'm suspecting the program code is being overwritten. This command should do the trick:

st-flash read backup.bin 0x08000000 0x20000

Dsl71, 2024-03-14:
[backup_boseanfs600_2_0_beta18.zip](https://github.com/Gissio/radpro/files/14600010/backup_boseanfs600_2_0_beta18.zip)

Here it is, its the same content than the original flash file.

Gissio, 2024-03-14:
First, thanks for your patience.

I am sending you beta19-test1.

[radpro-bosean-fs600-2.0beta19-test1-install.bin.zip](https://github.com/Gissio/radpro/files/14606811/radpro-bosean-fs600-2.0beta19-test1-install.bin.zip)

If the statistics view crashes, you should hear a series of pulses, after which the device should turn itself off. Please record these pulses using a voice recorder and attach them here to a message. They will let me know where the problem is.

Also, let me know if the settings are saved now.

Gissio, 2024-03-14:
First, thanks for your patience.

I am sending you beta19-test2.

[radpro-bosean-fs600-2.0beta19-test2-install.bin.zip](https://github.com/Gissio/radpro/files/14607495/radpro-bosean-fs600-2.0beta19-test2-install.bin.zip)

If the statistics view crashes, you should hear a series of pulses, after which the device should turn itself off. Please record these pulses using a voice recorder and attach them here to a message. They will let me know where the problem is.

Also, let me know if the settings are saved now.

Dsl71, 2024-03-15:
No Problem, thank you for your effort to bring trash counters to good counters!

Settings are not saved.

Entering the statistics screen, immediate crash, no pulses,no shutdown.

Sorry, i have no better news.

Gissio, 2024-03-15:
Beta20 (https://github.com/Gissio/radpro/releases) should fix the statistics screen crash.

Dsl71, 2024-03-15:
hmm statistics screen doesn't crash, values (temp between -7 to 125) seems to be not correct.
Settings are not saved.
deadtime shows -.-- or so.

Gissio, 2024-03-16:
I forgot to ask, does the battery indicator show the correct level of charge when unplugged?

Dsl71, 2024-03-16:
Alkaline battery, 3,9V , battery indicator shows 100% set as Ni-MH, about 30-40% set as Alkaline. But they are very weak already,have to get new ones.

Gissio, 2024-03-16:
Ok. I think I found the temperature problem: on the STM32F0x1 and STM32F1x3, temperature is read on ADC channel 18; now, on the STM32G07x, temperature is read on ADC channel 12.

So we're just left with the flash program problem. I'll keep you updated.

Gissio, 2024-03-16:
I just released beta21. The statistics view should show now the device battery voltage (per cell). Are settings preserved now?

Dsl71, 2024-03-17:
no good news for beta22 and beta23.
Battery voltage shown  around 1,5V, no settings saved.

Gissio, 2024-03-17:
Ok.

There goes beta23-test1:

[radpro-bosean-fs600-2.0beta23-test1-install.bin.zip](https://github.com/Gissio/radpro/files/14628807/radpro-bosean-fs600-2.0beta23-test1-install.bin.zip)

You should see a "FLASH_SR=" line in the status bar, followed by hexadecimal numbers.

Please turn on datalogging to 1 minute. The numbers should change. Let me know what numbers you see. Please wait for a minute. Every minute the datalog should attempt to write to flash. Do the numbers change? To what value?

Thanks.

Dsl71, 2024-03-18:
0x00000000 -> 0xc0000000 after enabling data logging of 1 minute.
0xc0000000 stays all the time (5 minutes runtime or so)

Gissio, 2024-03-18:
I think I'm onto something:

[radpro-bosean-fs600-2.0beta23-test2-install.bin.zip](https://github.com/Gissio/radpro/files/14637878/radpro-bosean-fs600-2.0beta23-test2-install.bin.zip)

Can you repeat the previous steps? And check if settings are saved now?

Dsl71, 2024-03-18:
0x00000000 -> 0x00000008 after enabling data logging of 1 minute.
0x00000008 remains all the time (5 minutes runtime or so)
Settings are not saved. After reboot, if only settings are changed value remains at 0x00000000.

FYI: I do all of this usb powered.

Gissio, 2024-03-18:
Ok, here is what is happening: the flash controller is reporting a PROGERR, which happens when attempting to write data to a memory location that was already written. I have no idea why this should occur. But at least I know what to look for.

Dsl71, 2024-03-18:
Hmm, maybe you have to erase the used sectors before writing to them?

Gissio, 2024-03-18:
Here's what I think is happening. 

The STM32G070 has flash memory with error correction, so there are invisible error bits. So, whenever a value is written to this flash memory, the error bits are also set (although they are invisible).

The Rad Pro "-install" firmware sets unused memory to 0xff, which is value of erased flash memory. Now, because of these error bits, the STM32G070 hardware detects that data has been written previously.

So, indeed, on the STM32G070 it is necessary to erase the flash memory again.

Let's try beta34-test3:

Do we agree that this is a very nasty bug?

Gissio, 2024-03-18:
Here's what I think is happening. 

The STM32G070 has flash memory with error correction, so it has invisible error bits. So, whenever a value is written to this flash memory, the error bits are also set (even though they are invisible).

The Rad Pro "-install" firmware initializes settings and datalog memory to the value of erased STM32 flash memory. But because of these error bits, the STM32G070 detects that data has been previously written (by the st-link.exe flash software), and, correctly, refuses to reflash memory.

So, indeed, it is necessary to erase the flash memory.

Let's try beta23-test3:

[radpro-bosean-fs600-2.0beta23-test3-install.bin.zip](https://github.com/Gissio/radpro/files/14641231/radpro-bosean-fs600-2.0beta23-test3-install.bin.zip)

Do we agree that this is indeed very nasty?

Dsl71, 2024-03-18:
I agree, a nasty  bug. But still no good news, settings are not saved, the FLASH_SR= line is missing too.

Gissio, 2024-03-18:
Same as before, with FLASH_SR:

[radpro-bosean-fs600-2.0beta23-test4-install.bin.zip](https://github.com/Gissio/radpro/files/14642634/radpro-bosean-fs600-2.0beta23-test4-install.bin.zip)

Dsl71, 2024-03-19:
0x00000000 -> 0x00000008 after enabling data logging of 1 minute.
0x00000008 remains all the time (5 minutes runtime or so)
Settings are not saved. After reboot, if only settings are changed, value remains at 0x00000000.

If before data logging was activated, value is now 0x00800008 when settings are changed.
Settings are not saved.

Gissio, 2024-03-19:
Odd.

Can you download the flash memory with st-link.exe read and check if there are any differences?

Gissio, 2024-03-19:
Can you try again?

[radpro-bosean-fs600-2.0beta23-test6-install.bin.zip](https://github.com/Gissio/radpro/files/14657617/radpro-bosean-fs600-2.0beta23-test6-install.bin.zip)

If it fails, can you download the flash memory with `st-link.exe read` and let me know if something changed, and how?

Dsl71, 2024-03-20:
Yippie! partly success!!! What happens:
Change settings,leave menu, power cycle -> settings are here.
Do another power cycle -> settings are default again.

Gissio, 2024-03-20:
The good news is that this is possibly due to an unrelated problem. In the STM32G070, settings were not memory aligned to 8-byte flash words. It should work now:

[radpro-bosean-fs600-2.0beta23-test7-install.bin.zip](https://github.com/Gissio/radpro/files/14669912/radpro-bosean-fs600-2.0beta23-test7-install.bin.zip)

Dsl71, 2024-03-20:
Well done Sir!

For the first time settings are safed !!!!! Cudos, Gissio

Gissio, 2024-03-22:
I'm glad it works now.

I've just released rc1 (release candidate 1). Can you check if datalogging works with the Bosean FS-600?

I've also added compatibility with GeigerLog. Does this work: https://github.com/Gissio/geigerlog-radpro ?

Dsl71, 2024-03-22:
`   flash_unlock();

    for (uint32_t i = 0; i < size; i += FLASH_WORD_SIZE)
    {
        if (!flash_program(dest + i, source + i))
        {
            if (!flash_erase_page(iterator->pageIndex))
                break;

            if (!flash_program(dest + i, source + i))
                break;
        }
    }

    flash_lock();
`

Ah i see the new flash code, looks clean and neat. I will flash rc1 and report back.

Dsl71, 2024-03-22:
Ha a first connect and data collect with geigerlog-radpro works :-)

Dsl71, 2024-03-22:
First datalogging history...
[image]

Dsl71, 2024-03-22:
[image]

Dsl71, 2024-03-22:
[image]

Dsl71, 2024-03-22:
[image]

Gissio, 2024-03-22:
Regarding your first question: data logging stores cumulative pulse count. From there both rate and dose can be calculated.

Regarding the problem with datalogging: try the new geigerlog-radpro release. It shouldn't crash when requests don't get answered in time (there is a 250 ms timeout when the serial port is opened).

Dsl71, 2024-03-22:
Regarding: https://github.com/Gissio/radpro/issues/17#issuecomment-2014498168

I Think there should be some error handling too in the future. Otherwise you never know when a flash error occurs.

Dsl71, 2024-03-22:
[image]

I am sorry. Note: data logging of 1 minute is also active.

Gissio, 2024-03-22:
Can you try geigerlog-radpro 1.4.3b?

I've changed the Rad Pro flash code once again. Now a page is only erased if writes fail at the start of the page. This improves recovery in case of an error.

Dsl71, 2024-03-23:
geigerlog-radpro 1.4.3b runs for one hour now without any problems. 

Next step: Flash the newest FW and retry...

Dsl71, 2024-03-23:
Bug, see the two timestamps...

[image]

Dsl71, 2024-03-23:
Bug , from a certain point "no data", no recovery without reconnect

[image]

Dsl71, 2024-03-23:
This is also very very unlikely:

[image]

Dsl71, 2024-03-23:
Again no data, device freezes, only pulse led is updated.

[30 more replies]

closed 2024-04-21 by Gissio