Gissio/radpro #20: Bricked!

morganatOmmo, 2024-03-08:
I just tried to update from the original vendor code to the latest update (I didn't see the note about using the initial version, before updating, until it was too late).

Sadly, my unit is now bricked and won't instantiate a UCB drive to allow further updating.

My device uses the CH32F103 MCU, from WCH, not the GD one - is this a direct equivalent?  Will I be able to program the image that I should have initially programmed onto the device via the SWD interface?

Thanks!

Gissio, 2024-03-08:
If you did not tinker with the device at a hardware level, I don't think you need to reflash the bootloader.

Some users mentioned that they had to press the right/settings key before pluggin in USB. Have you tried this, or tried other key combinations?

If that doesn't work, open your device, unplug the battery connector from the main board, remove the button battery, and try plugging in USB.

morganatOmmo, 2024-03-08:
Hi Gissio, 

Thank you for the feedback.  I haven't touched the FW or HW, its as standard.

Without the batteries connected I can't even get the bootloader.

With the main battery disconnected and the memory backup battery removed, I can get the bootloader to bring up a drive, but when I try to copy the image onto it, it reports that the drive is too small and that I need an additional 8K to copy the file.

I also tried all of the various key combos.

Well, I guess that is it for this device...

Gissio, 2024-03-08:
I'm sorry but this forum is dedicated to Rad Pro, not to general geiger counter issues. I suggest getting help in specialized forums like https://www.radmon.org, for instance.

Gissio, 2024-03-08:
Try flashing V1.5!

Gissio, 2024-03-08:
You can get it here: https://github.com/Gissio/radpro/tree/main/docs/devices/FNIRSI%20GC-01/firmware

morganatOmmo, 2024-03-08:
Thanks, I have my GC-01 working again!
I did a GM tube upgrade and added clicks in a couple of little videos on my YT channel:
https://www.youtube.com/channel/UCSTS7Jg1eavY02riCO63yPg
Maybe, you can add some more energy to the LED flashes to help increase the volume of the clicks that are generated?  Then I can make a video about that too, highlighting your additions to Amazon's top selling counter.
One question, how come you didn't get rid of the leading zeros on the graph?

Gissio, 2024-03-08:
I'm glad you fixed it. In short, Rad Pro will be available for GC-01. Til then, stay tuned.

Gissio, 2024-03-08:
I'm glad you fixed it. In short, Rad Pro will be available for GC-01. Til then, watch the GitHub project so you get notified when it is ready.

Gissio, 2024-03-14:
Rad Pro 2.0beta19 for the FNIRSI GC-01 has been released! You can find it here: https://github.com/Gissio/radpro/releases

Remember it is beta software, so there is room for improvement.

morganatOmmo, 2024-03-16:
This works on my GC-01.  Would you object to me adding this FW to a updated review video on the GC-01 product?

 

Thanks!

Gissio, 2024-03-16:
It would be an honor.

But please consider the following things when reviewing Rad Pro for FNIRSI GC-01:

* Please (!) make it clear that a Windows computer is **necessary** to install Rad Pro on the FNIRSI GC-01. The bootloader **does not** support macOS/Linux. I wouldn't want to receive a flood of complaints because this is not clear.
* Please make it clear that Rad Pro is currently in beta. It would also be nice if you tell people to subscribe to the repository so they get notified of new versions.
* Maybe you find this fact interesting: Rad Pro's user interface is based on a design system used in the maritime industry (www.openbridge.no). That means that all the work from the OpenBridge team makes it easy to use Rad Pro in field conditions.
* Have you been able to try the data communications to download the datalogs?

morganatOmmo, 2024-03-16:
Hi Gissio,

 

I will be sure to mention the points you brought up, please also understand that I am very honest in y reviews, my little channel is not commercial, so that kind of means I get to say what the f*ck I like, but I do like these changes in general.

 

I have not been seen any logging, I saw this option in the menu system, but did not see any log files etc, what should I be looking for?

 

FYI, my channel on YT can be found here:

https://www.youtube.com/channel/UCSTS7Jg1eavY02riCO63yPg

 

Kind Regards

Gissio, 2024-03-16:
I like honest. But also well-intentioned. If you find a flaw, I would appreciate reporting it here before telling your audience.

Forget about the logging for now.

morganatOmmo, 2024-03-16:
I will be a little while before I get around to the video, I have a backlog, but please let me know when the logging function is available, this would be a really major feature to be able to discuss.

 

I will be sure to keep you in the loop on my findings, normally, I make an unlisted release of my videos first, so that the vendors have an opportunity to respond before the final cut.

 

Take care...

 

Forget about the data logging for now.

Gissio, 2024-03-16:
Data logging should work right now. It just happens that nobody tested it so far.

So, if you would like to be the first betatester, here are the steps:

* In Rad Pro, go to `Settings`, choose `Data logging` and select the data logging interval of your liking. Wait some time to get some data.
* Go to Windows, open the `Device Manager`.
* With your device on, connect it to your computer. In the `Device Manager` a new `USB Serial Port` should appear under `Ports (COM & LPT)`. Take note of the COM port.
* Download the radpro-tool (it's in the releases: https://github.com/Gissio/radpro/releases)
* Open a terminal (cmd.exe).
* Run this command:

    [path to radpro-tool-1.0.exe] --help

This should show you all the options.

For example, to download the datalog, you need to type:

    [path to radpro-tool-1.0.exe] -p COMx --download-datalog [filename]

Where `COMx` is the COM port you took note of previously, and `[filename]`, the filename with the downloaded datalog. It is best to name it with `.csv` extension, so you can open it in any spreadsheet editor.

Please keep in mind that these tools are intended for people with some programming skills. Perhaps you can convince the people at https://sourceforge.net/p/geigerlog/wiki/Home/ to add support for Rad Pro (should make downloading and processing datalogs much easier).

Gissio, 2024-03-16:
Data logging should work right now. It just happens that nobody tested it so far.

So, if you would like to be the first betatester, here are the steps:

* In Rad Pro, go to `Settings`, choose `Data logging` and select the data logging interval of your liking. Wait some time to get some data.
* Go to Windows, open the `Device Manager`.
* With your device on, connect it to your computer. In the `Device Manager` a new `USB Serial Port` should appear under `Ports (COM & LPT)`. Take note of the COM port.
* Download the radpro-tool (it's in the releases: https://github.com/Gissio/radpro/releases)
* Open a terminal (cmd.exe).
* Run this command:

    [path to radpro-tool-1.0.exe] --help

This should show you all the options.

For example, to download the datalog, you need to type:

    [path to radpro-tool-1.0.exe] -p COMx --download-datalog [filename]

Where `COMx` is the COM port you took note of previously, and `[filename]`, the filename with the downloaded datalog. It is best to name it with `.csv` extension, so you can open it in any spreadsheet editor.

Please keep in mind that these tools are intended for people with some programming skills. Perhaps you can convince the people at https://sourceforge.net/projects/geigerlog/ to add support for Rad Pro (should make downloading and processing datalogs much easier).

morganatOmmo, 2024-03-19:
The link you sent to the RadPro tool seems to link to the device firmware, not PC code.  Did I misunderstand something?

Gissio, 2024-03-19:
Hello there,

I'm sorry but I forgot to notify you that we've found a bug in data logging. I am in the process of fixing it.

Once it is ready, I'll let you know.

morganatOmmo, 2024-03-20:
No problem.

 

A couple of questions:

 

1.	What font does the RadPro use?
2.	Would you like me to include any kind of Bio for you in the video?

 

Thanks!

Gissio, 2024-03-20:
The font is Roboto (https://fonts.google.com/specimen/Roboto?query=roboto).

A reference to Gissio is more than enough!

morganatOmmo, 2024-03-20:
Wow, the power consumption is pretty high...

Gissio, 2024-03-20:
Indeed.

morganatOmmo, 2024-03-21:
https://youtu.be/23sX-_Oo_h0

 

unlisted version of the video...

Gissio, 2024-03-21:
I think your video captures the essence of Rad Pro.

Can you hold a day or two? Data logging with a proper data logging software is almost ready!

morganatOmmo, 2024-03-22:
For the datalogging, I was thinking of a follow up video anyway.  I would prefer to wait for that feature to be robust and also have a simple UI.  Command line operation is going to be asking a lot from the average folk.

 

The FNIRSI GC-01 is not the only counter that you support, so possibly a later video would include a wider discussion of your project.

 

The video is scheduled for release 11pm on Saturday night (Beijing time 23rd March)

 

Thanks!

Gissio, 2024-03-22:
That sounds great.

Btw, I just finished adapting GeigerLog to Rad Pro. I will probably release the next version tomorrow.

You can read more about GeigerLog here:

https://sourceforge.net/projects/geigerlog/

The user interface is not very good, but the program does its job, and it also features interesting data processing tools: Poisson histogram fit, FFT regularity analysis and custom averaging, among others.

morganatOmmo, 2024-03-22:
Hi Gissio

 

It all sound really cool.  Once the code is ready and mature, I will start on a new video.

 

Perhaps you can tell me a little about yourself, for example, in my video, I kept everything gender neutral, I know nothing about you...

 

What drives you create this firmware? How did you get started in engineering?  Do you have any stories to tell?

 

Thanks!

 

Morgan

 

https://sourceforge.net/projects/geigerlog/

Gissio, 2024-03-26:
A question: in your video you mentioned that the battery lasts only 24 hours.

I found this actually surprising, as I do two things the original firmware doesn't:

* The MCU is put in sleep mode whenever possible, reducing average power consumption by 15 mA.
* The display is put in sleep mode when it it turned off (the original firmware would only turn it off), reducing average power consumption by 10 mA.

Could you install the original firmware, max out display brightness and measure the current at the battery? Could you wait for the display to go to sleep and measure again? Could you repeat these measurements with Rad Pro?

By the way, there was a power problem with beta23: the device would continue draining the battery even if it was turned off (this was fixed in 2.0rc1). But I think this is unrelated to what you are reporting.

Gissio, 2024-03-26:
A question: in your video you mentioned that the battery lasts only 24 hours.

I found this actually surprising, as I do two things the original firmware doesn't:

* The MCU is set to sleep mode whenever possible, reducing average power consumption by 15 mA.
* The display is set to sleep mode when off (the original firmware does not use a power saving mode), reducing average power consumption by 10 mA.

Could you install the original firmware, max out display brightness and measure the current at the battery? Could you wait for the display to go to sleep and measure again? Could you repeat these measurements with Rad Pro?

By the way, there was a power problem with beta23: the device would continue draining the battery even if it was turned off (this was fixed in 2.0rc1). But I think this is unrelated to what you are reporting.

jonwhite227, 2024-04-05:
Hi, I seem to have bricked my GC-01. I followed the steps in Studio 326's video, but as soon as I dragged the firmware file into the USB window the display went off, never to return. 

I have tried all the suggestions on here, and from Studio 326 - re-installing V1.05 of the firmware, installing newer version of firmware (checking that the processor etc. is correct), resetting the unit with the reset switch and centre button pushed down, disconnecting both batteries, but I have not had a peep out of it since. I can get the USB window to popup but that is all. If I reconnect the battery and connect to USB, I get a message 'the device connected has malfuntioned'. I can disconnect battery and do it all again, but it just goes round in circles with nothing appearing on the display. Have tried USB 2 and USB 3 connection and holding/not holding centre button while updating.

The GC-01 was bought in November 2023 and was running 1.6-2 firmware and a J364 tube, which I have swapped out with a J321 tube.  It was working fine, if a little over sensitive with the new tube, displaying around 3x background, compared to my 'proper' yellow July 2022 model (0.45uSV instead of 0.14uSV.

Does anyone have any ideas to bring it back to life, or have I got a spare J321 tube and a nice blue paperweight? Any suggestions are very welcome!

Gissio, 2024-04-05:
Hi Jon,

In a few days I'll post instructions for recovering FNIRSI GC-01 devices. In the meantime, get an [STLINK](https://www.amazon.com/s?k=stlink) dongle and a 4-pin header!

Gissio, 2024-04-05:
Hi Jon,

what OS does your computer have? What MCU does your GC-01 have? According to many user reports, the GC-01 requires a Windows computer for the upgrade to work.

In case you've already tried from Windows and were not able to flash any firmware: in a few days I'll publish instructions for recovering FNIRSI GC-01 devices. For this to work, you'll need an [STLINK dongle](https://www.amazon.com/s?k=stlink) and a 4-pin header.

jonwhite227, 2024-04-05:
Hi, thanks for your reply. I am running Windows 10 and my GC-01 has the CH32F103R8T6 MCU. 
Is the 4 pin header male or female (pins or socket)?

dc1rdb, 2024-04-05:
Hi Jon, that depends on your cable to the ST-Link. I've soldered a male header onto my board.
[image]

Gissio, 2024-04-05:
Jon:

can you go to "Security" and draft a new "Security Advisory"? Send me your email address so I can contact you. Your email address won't be published.

Gissio, 2024-04-21:
Closed as all issues have been solved.

closed 2024-04-21 by Gissio