Gissio/radpro #21: FS2011 (GD32F150) battery and temperature readouts wrong on main branch

mryndzionek, 2024-03-10:
The temperature shows >300C and the battery icon shows "charging" all the time.

Gissio, 2024-03-10:
Another regression... could you also please check these?

- Dead-time measurement works (takes some hours).
- Random generator works.
- Display contrast/brightness settings work.
- Device clock works.
- radpro-tool works (over SWD).
- Data logging works (read datalog with radpro-tool).

mryndzionek, 2024-03-10:
- &#x2611; Display contrast/brightness settings - works
 - &#x2611; Device clock - works
 - &#x2612; Random generator - doesn't seem to be working

Gissio, 2024-03-10:
Can you try again?

[radpro-fs2011-gd32f150c8-2.0beta17-test1-install.bin.zip](https://github.com/Gissio/radpro/files/14552073/radpro-fs2011-gd32f150c8-2.0beta17-test1-install.bin.zip)

The statistics view will not display the dead-time, but rather two hexadecimal numbers separated by ":". Both should change. Let me know if they do not change, if only one changes (which one) or if both change). Also, if both change, let me know which one changes faster.

mryndzionek, 2024-03-11:
> Can you try again and see if the ADC works?
> 
> [radpro-fs2011-gd32f150c8-2.0beta17-test1-install.bin.zip](https://github.com/Gissio/radpro/files/14552073/radpro-fs2011-gd32f150c8-2.0beta17-test1-install.bin.zip)
> 
> The statistics view will not display the dead-time, but rather two hexadecimal numbers separated by ":". Both should change. Let me know if they do not change, if only one changes (which one) or if both change. Also, if both change, let me know which one changes faster.
> 
> pd: each number represents the count of a 16-bit timer. They should chain to form a 32-bit timer. If the timers do not work, there is no dead-time measurement nor random number generation.

ADC still doesn't work and both dead-time values are always zero.

Gissio, 2024-03-13:
I just released Rad Pro 2.0beta18. Can you go over this checklist?

- The battery indicator works correctly.
- The temperature indicator in the statistics menu shows reasonable values.
- The random generator and dead-time compensation work.
- The serial port communication works.
- Data logging works and you can download it with radpro-tool.

mryndzionek, 2024-03-13:
Random generator seems to be working. Dead time shows some (I think sane) values decreasing with time. The rest without changes.

Gissio, 2024-03-13:
So the ADC is still not working...? I'll keep you updated.

mryndzionek, 2024-03-13:
> So the ADC is still not working...? I'll keep you updated.

Yeah, the temp. shows 378.7C and the battery icon is "charging".
Don't feel pressured. I'll wait :smile: .

Gissio, 2024-03-13:
I'm not.

Gissio, 2024-03-13:
Ok. Found it.

[radpro-fs2011-gd32f150c8-2.0beta18-test1.zip](https://github.com/Gissio/radpro/files/14594641/radpro-fs2011-gd32f150c8-2.0beta18-test1.zip)

mryndzionek, 2024-03-14:
> Ok. I think I found the error. Can you try again? Try beta19.

Yes, now it works. The temperature is a little bit high (37.8C), but yeah, the internal sensor is probably not that accurate.

Gissio, 2024-03-14:
Excellent! Remember device temperature is MCU temperature, not ambient temperature, so the value looks great.Ok. I think I found the error. Can you try again? Try beta19.Yes, now it works. The temperature is a little bit high (37.8C), but yeah, the internal sensor is probably not that accurate.

closed 2024-03-14 by Gissio