Gissio/radpro #25: FNIRSI GC-01: Turns on in HOLD mode

martin-machacek, 2024-03-14:
Everytime I turn the device on, it is in the Hold mode of Instantaneous screen. It does not matter if I previously turned it off in Hold mode or not. While turning it on, I release the center OK key as soon as the LED turns off and Rad Pro splashscreen appears.

martin-machacek, 2024-03-14:
I may be holding the key for too long. If I release the key sooner, it still turns on, but not in Hold mode.

Gissio, 2024-03-14:
I'm glad you could install Rad Pro successfully on your device!

Regarding the issue, this can be fixed! Thanks for the report.

Tig3rch3n, 2024-03-15:
Just confirmed the issue.
If you hold Power just 1 Sec, Instantaneous is NOT on Hold
If you hold Power for >2 Sec it starts on Hold

Gissio, 2024-03-15:
Beta20 should fix this. Let me know.

martin-machacek, 2024-03-16:
I can confirm that it is fixed 👍

Gissio, 2024-03-16:
Nice!

closed 2024-03-16 by Gissio