Gissio/radpro #28: FNIRSI GC-01: wrong device temperature statistics

Alter0ne, 2024-03-16:
Hello!
I have the device with ch32f103 processor. It shows the incorrect temperature: 9 °C when the environment is 28 °C.
[Photos](https://imgur.com/a/HX86Gs4)
Also my device resets the current date and time after rebooting. But the time zone is saved correctly. Perhaps it's my battery problem?

Gissio, 2024-03-16:
Regarding the real-time clock. The date and time should work correctly. If it doesn't, check the button battery's voltage (it should be a CR1220). It should be around 3 V. If it doesn't, replace it!

And the temperature discrepancy is expected. STM32F103 and compatible microprocessors are not temperature calibrated in the factory. This means there can be a variation of up to ± 20 °C from the actual value, because the temperature sensor of different production batches can be phyiscally different.

To fix this, there should be a setting to calibrate this. That's actually a question I have been asking mytself. And this question goes to everyone who reads this. Does it make sense to show uncalibrated temperature? Should there be settings to fix this? In order to keep things simple, I think it would be best to not show temperature at all (if it's wrong, it makes no sense as it confuses more than helping).

martin-machacek, 2024-03-16:
My device shows the temperature 13 to 15 degrees Celsius higher (ambient temp is 24, GC shows 38). In my opinion it doesnt make sense to show it, if there is no way to calibrate.

Gissio, 2024-03-16:
Agreed. Next beta will replace device temperature with device battery voltage (which makes more sense).

Gissio, 2024-03-16:
I just released beta21 which fixes this.

https://github.com/Gissio/radpro/releases

martin-machacek, 2024-03-16:
Unfortunately it turns on and the screen is blank (probably the background color, does not seem to be white). I can hear particle clicks and particle LED flashes, but nothing else. Key presses don't help.

Gissio, 2024-03-16:
My bad. Try beta22 now!

martin-machacek, 2024-03-16:
beta22 works and shows the battery voltage. 
4.539V connected to USB, after disconnecting it slowly settles for cca 10 minutes down to 3.780 and than slower as it discharges.

Gissio, 2024-03-16:
Nice!

Gissio, 2024-03-16:
Can I make a request? It would be great if you can measure the discharge curve of the battery so the battery indicator shows a correct value.

This would imply fully charging the device, and taking note of the voltage value every 10 minutes. It would be great if you do this with the screen off. Can you do this?

martin-machacek, 2024-03-16:
I can try tomorrow. I am thinking of ways to automate this. Wouldn't it be possible to log it internally?

Gissio, 2024-03-16:
It would be, but I think it takes less time to do this manually than for me to program the code.

martin-machacek, 2024-03-16:
Ok, I will give it a try tomorrow then

Tig3rch3n, 2024-03-16:
I can test both of my devices with the power profiler in a few days.
I even plan to do different Setups

Brightness max/Display on all the Time vs Brighness min/Sleep after 30sec and a few in between

Gissio, 2024-03-16:
A measurement on one device should be enough. 

Use only brightness max/display on all time so the measurement takes less time.

Thanks in advance!

Tig3rch3n, 2024-03-17:
Btw... would it be possible to enable a Temp Offset?
I have a big'ish diff in both directions

Roomtemp 23°
Device 1 14°
Device 2 27.5°

martin-machacek, 2024-03-18:
Temp offset setting would be nice.

I am still collecting the voltage values. The battery lasted whole day yesterday and still has 3.533V. After having it switched off for the night I have charged it up for 30s and when the voltage dropped back to the last value from yesterday, I have resumed logging.

Tig3rch3n, 2024-03-18:
Same, i did a Sleep after 30s run yesterday and do a always on min brightness at the moment :)

martin-machacek, 2024-03-18:
Here is my voltage log: [Fnirsi-GC-01-voltage-log.txt](https://github.com/Gissio/radpro/files/14636246/Fnirsi-GC-01-voltage-log.txt)
The device turned off after the last value of 3.042V
Parameters during test:

- Interval 10 minutes
- Display sleep after 30 seconds
- Display brightness level High 
- Rad Pro version 2.0beta22

Gissio, 2024-03-18:
They say an image is worth a thousand words.

[image]

Thanks for all the hard work!

Gissio, 2024-03-22:
Updated curve is available in 2.0rc01!

closed 2024-03-22 by Gissio