Gissio/radpro #29: Fnirsi downloadin error

destrudos, 2024-03-16:
Thank you for the great work with radpro!
However I suffer from issue with downloading data from fnirsi (both datalog, and in real time)

`python tools/radpro-tool.py -p COM3 --no-sync-time   --download-datalog moj-datalog.csv
Downloading offline data...`

Gives no error but moj-datalog.csv is empty, only the:

`time,tubePulseCount` are present, also:
`python tools/radpro-tool.py -p "COM3"  --log-rng odczyty.csv`
Gives empty file. Of course logging option is enabled in the device.
Also sometimes, I encounter other strange behavior, for example:
`python tools/radpro-tool.py -p COM3 --get-device-id`
`Error while setting device time.`
`Error while getting property "deviceId".`
`(base) PS E:\PROGRAMMING\radpro> python tools/radpro-tool.py -p COM3 --no-sync-time   --download-datalog moj-datalog.csv`
`Downloading offline data...`
`Error while downloading data log.`
  What is the reason of this?

Gissio, 2024-03-16:
Thanks for testing data logging!

If the datalog shows as empty, could it be that there hasn't been anything logged yet?

Try the following command:

    pyserial-miniterm.exe

And connect to the GC-01's serial port.

Press enter. You should receive an `ERROR` response.

Try the commands listed here: https://github.com/Gissio/radpro/blob/main/docs/comm.md . In particular, the `GET datalog`. Make sure that upper and lowercase are correct.

Let me know.

Gissio, 2024-03-17:
I have been able to replicate the error! Thanks for the report.

Gissio, 2024-03-22:
I've just released rc1 (release candidate 1). Can you check if it works now?

I've also added compatibility with GeigerLog. Check this out: https://github.com/Gissio/geigerlog-radpro

Gissio, 2024-04-21:
Closed as it seems to be working now.

closed 2024-04-21 by Gissio