Gissio/radpro #30: FNIRSI GC-01 Cannot enter bootloader

hackra76, 2024-03-19:
Hey guys, I have a problem, I bought a FNIRSI GC-01 on latest firmware v1.6-2, with APM32F103RBT6 (Geehy) microcontroller. Picture of motherboard included.
I wanted to try the radpro firmware, but I am stuck on accessing the drive. If I follow the install instructions, I turn off the device,
plug a USB C cable (verified data transfer cable) into the device and pres the power button, the device just boots up and starts working.
No drive shows up in My computer, no device connected beep is heard. The machine is a Lenovo Legion 5i laptop running latest Windows 11. I am out of ideas, I tried holding down right button while plugging the USB, I tried disconnecting the internal battery and removien the backup battery, nothing works.... Please help me.... What am I doing wrong ? What could I try ?

[image]

martin-machacek, 2024-03-19:
I have the same issue, although I have Windows 10. **It works only when I use a USB hub.** I have heard that it needs USB 2.0 to work and my laptop has only USB 3 ports. However the USB hub is USB 3 too, but it works.

JantarX, 2024-03-19:
Hello. If you have a device from the manufacturer on version 1.6 or higher, it is possible that you will have to invoke the bootloader on the turned off device with the settings button (right arrow) + power button at the same time while inserting the USB cable into the PC. ... and yes, you must have a USB 2.0 input to the PC. I can't get to you on my laptop with USB 3.0 either.

hackra76, 2024-03-19:
I just had the craziest idea based on your reply... I have no access to USB2.0 port now, but I have a Samsung Galaxy S23ultra... So I grabbed a USB-C to USB-C cable, plugged one end into my Samsung device, the other into GC-01 and hit power button ...
Magic happened, the GC-01 booted to bootloader, the phone said something about unsupported STM drive and offered to format the connected drive. I refused that, unplugged the GC-01 from my phone an plugged it into a standard USB3.0 port on my laptop. And voila ! New drive appeared in My computer and I was able to copy the firmware file to the device. The GC-01 rebooted itself right after copying has finished and now it is running radpro !!!!

martin-machacek, 2024-03-19:
I have also encountered a problem when I connected it to my computer and than unplugged it without uploading any firmware. It looked like it erased the program memory and just did not work until I uploaded some firmware (FNIRSI or Rad Pro), so don't be surprised when it happens.

JantarX, 2024-03-19:
> I have also encountered a problem when I connected it to my computer and than unplugged it without uploading any firmware. It looked like it erased the program memory and just did not work until I uploaded some firmware (FNIRSI or Rad Pro), so don't be surprised when it happens.

Then you need to press the reset button with a needle, which is on the right side at the bottom of the device.

Tig3rch3n, 2024-03-19:
With the original Firmware there is a super weird Time Window for the USB Storage mode

If you hold Power for to short... nothing happens
If you hold power for just right... the USB Storage appears
If you hold Power to long... it just boots the current Firmware

RadPro is not so Picky for some reason o_O

Gissio, 2024-03-19:
> With the original Firmware there is a super weird Time Window for the USB Storage mode
>
> If you hold Power for to short... nothing happens If you hold power for just right... the USB Storage appears [...]

With the device turned off, doesn't the bootloader start automatically when you connect USB?