Gissio/radpro #37: FNIRSI GC-01: Battery icon stays the same not depending the battary charge

mileshkin, 2024-03-22:
Battery icon stays the same not depending the battary charge percent or charging process.
It shows fully gray battery with gray flash icon on it.

Gissio, 2024-03-22:
Go to the statistics menu. What's the battery voltage?

mileshkin, 2024-03-22:
It's 3.85 V

Gissio, 2024-03-22:
Strange, everything seems nominal.

Can you attach a picture of the statistics screen?

mileshkin, 2024-03-22:
[image]

Gissio, 2024-03-22:
You didn't mention the flash symbol!

It seems the GC-01 thinks it's charging, while it's not. I'll report once I have news.

martin-machacek, 2024-03-22:
He did:
> Battery icon stays the same not depending the battary charge percent or charging process. It shows fully gray battery **with gray flash icon on it**.

I have encountered this issue, too.

mileshkin, 2024-03-22:
> You didn't mention the lighting symbol!
" with gray flash icon on it " was exactly about it... may be my bad english caused you missunderstood me

Gissio, 2024-03-22:
I'm sorry, it seems the cold I'm going through is worse than I thought.

Gissio, 2024-03-22:
Can you try 2.0rc2?

mileshkin, 2024-03-23:
> Can you try 2.0rc2?

lighting symbol gone... at least. Battery is fully charged for now and the indicator shows full battery.

Gissio, 2024-03-23:
Unfortunately the GC-01 hardware signals "charging" when it is not plugged to USB. So there is no easy way to distinguish from: "unplugged", "charging on USB", and "plugged-in to USB with batteries charged".

closed 2024-03-23 by Gissio