Gissio/radpro #43: Ok Guys, here is the foolproff to get the FW loaded. Works everytime.....

jackhen, 2024-03-24:
Ok Guys, here is the foolproff to get the FW loaded. Works everytime.....
After you connect your GC-01 via usb to your PC you "hold down" the power switch. As long as you hold it down, you will see the disk appear on your PC. So hold it down, then drag the repo onto your disk icon. Release the power button when you see the GC-01 booting. Takes only a few seconds. And it doesn't matter the USB protocol. It just works.

_Originally posted by @jackhen in https://github.com/Gissio/radpro/discussions/24#discussioncomment-8890588_

Gissio, 2024-03-24:
Added to the installation instructions. Thanks for the report.

closed 2024-03-24 by Gissio

jackhen, 2024-03-24:
Gassio,
You can go to
https://www.youtube.com/watch?v=23sX-_Oo_h0&lc=UgzLD925mnA5s3N0Fzp4AaABAg.A1Lv1ezGxPDA1MFH59vPFe
and see this guy and what he shows based on your FW. You can also see his
technique to load your FW.