Gissio/radpro #46: F-Nirsi: At least 50% more battery consumption 'data logging' is activated

w01fdev, 2024-03-25:
I used the 'data logging' function with 60 minutes today. I noticed that the f-nirsi ran down the battery at least 50% faster than without the function.

Presumably only a short text line or file with the values should be made every 60 minutes, right?

I have now deactivated the function and will check the time without 'data logging' again after an hour of charging.

Gissio, 2024-03-26:
¿Could you measure the current with a multimeter?

w01fdev, 2024-03-27:
> ¿Could you measure the current with a multimeter?

No, unfortunately not. I'm not familiar with hardware measurements, but it's probably because of this function. After I deactivated it, the time was ~%50 higher with battery.

Is it not simply possible that a CSV is created every x minutes and can be retrieved when you connect the fnirsi to the PC (as if you were installing the firmware) This should only consume a minimal amount of energy.

1MegaSaxoN, 2024-04-14:
does anyone know if someone has tried to upgrade the battery in these gc-01"s yet. i an interested if anyone has input!

Gissio, 2024-04-14:
I just measured power consumption with and without datalogging. It is virtually identical.

Why do you say "that the f-nirsi ran down the battery at least 50% faster than without the function"?

> No, unfortunately not. I'm not familiar with hardware measurements, but it's probably because of this function. After I deactivated it, the time was ~50% higher with battery.
>
> Is it not simply [...]

w01fdev, 2024-04-15:
> I just measured power consumption with and without datalogging. It is virtually identical.
> 
> Why do you say "that the f-nirsi ran down the battery at least 50% faster than without the function"?
> 

Quite simply: because in my case it was most likely due to the fact that the fnirsi used about ~50% more battery for this reason, as I can't remember having adjusted anything else. But I am of the opinion that I only changed that.

Of course, I only did this once, so I don't know if it's reproducible. I will do this again by Wednesday at the latest with and without the function activated and also of the volt display and then take pictures.

Maybe there is someone else here who can test this?

My procedure:
1. wait until the battery is almost empty
2. Charge for 60 minutes
3. note the volts with statistics
4. let it run until almost empty
5. take pictures of time and volts

dc1rdb, 2024-04-15:
On my GC-01 with Geehy processor, enabling data logging has zero effect on the current consumption (measured with Fluke 115).

Gissio, 2024-04-15:
> Quite simply: because in my case it was most likely due to the fact that the fnirsi used about ~50% more battery for this reason, as I can't remember having adjusted anything else. But I am of the [...]

I think the problem is your measurement procedure.

> I read in a German forum that a 2000 battery should also fit in terms of size. But I'm not sure whether this has really been tried out or whether it was just an assumption.

Can you post a link?

w01fdev, 2024-04-15:
> > I read in a German forum that a 2000 battery should also fit in terms of size. But I'm not sure whether this has really been tried out or whether it was just an assumption.
> 
> Can you post the link?

https://www.geigerzaehlerforum.de/index.php/topic,1312.msg15608.html#msg15608

closed 2024-04-15 by Gissio