Gissio/radpro #48: New mcu problem CA32F0502L3

yo8saf, 2024-03-27:
hi, congratulations on your work. I have a big problem, I bought the cg-01 from China and it arrived with the 613-614 tube, it doesn't write. I ordered the j321 tube to change it. My mcu is a 
[image]
[image]
[image]. I've tried every way and I can't get it to show on the computer. can you help me thanks. are you a radio amateur? I'm YO8SAF Aldo
i use google transate sorry but  I use google translator, because my English is not very good, sorry

Gissio, 2024-03-27:
I can't support an MCU for which there is no publicly available datasheet (http://www.cachip.com.cn/Products_table/2.html).

yo8saf, 2024-03-27:
so nothing can be done?

Gissio, 2024-03-27:
No.

Gissio, 2024-03-27:
Actually, this looks like a copycat. Also, the USB data pins seem to be disconnected, so there is no way it could show up as a USB drive. Maybe you can return the device and demand a "genuine" GC-01 with a J321 tube?

dagaj, 2024-04-01:
hi, There is MCU CA32F0602L3-19 (Turn off the device. Connect the charger. Press and hold the up arrow key. The processor model will be displayed.) The tube used is not j613/j614. The tube parameters are: Effective length: 60 mm Max. length: 88 mm Diameter: Ø 9.2 mm

danielskowronski, 2024-04-06:
Got same issue - same MCU, same tube. After closer inspection, there's plenty of evidence for such device being fake. 

Some other signs include:

- instructions that omit Fnirsi logo (otherwise 1:1 copy of PDF from manufacturer)
- UI with minor differences (like `lifetime` vs `runtime`, which contradicts attached manual)
- starting value always `000.07 uSv/h`
- ability to select languages other than English or Chinese (like Polish)
- label on back being 1:1 copy with QR code leading to proper manufacturer website, but with `http://www.fnirel.cn/` printed below...