Gissio/radpro #57: FNIRSI GC-01 cricked firmware before update

Red-Owl, 2024-04-07:
i have rewrite a cpu firmware ad have used https://www.amazon.it/dp/B07H9XV2W2/ to upload firmware in jtag, check wire connection from to gg-01, the cpu is ch32f103r8 che st v2 read the cpu, and succefull write, the display remain blank

i use a stm32 official software in windows to write a cpu.

Gissio, 2024-04-07:
Attendi un paio di giorni, pubblicherò istruzioni per ripristinare il GC-01 con un dongle STLINK!

Red-Owl, 2024-04-12:
io aspetto :)

Gissio, 2024-04-21:
Ecco le istruzioni per il flashing dell'FNIRSI GC-01 con un STLINK: https://github.com/Gissio/radpro/blob/main/docs/devices/FNIRSI%20GC-01/install-stlink.md

Red-Owl, 2024-04-21:
Upload facilissimo, 
veramente ottimo fa tutto da solo.

[image]
[image]
[image]

che comando devo usare per leggere i dati di continuo dal pc ?
posso chiederti se puoi mettere i comandi in json tipo:

[{"cmd":"getdeviceId"}]
...
[{"staus":"OK","model":"FNIRSI GC-01 (CH32F103C8)","sw":"Rad Pro 2.0rc3;8869569e"}]

dato che lo leggo da javascript basta fare un **eval o json parse** e restutuisce un array facilmente utilizzabile
....
(nodejs su nw.js)

Gissio, 2024-04-21:
Eccellente!

Leggi https://github.com/Gissio/radpro/blob/main/docs/comm.md .

closed 2024-04-21 by Gissio