Gissio/radpro #66: FS2011 Flashing and disapearing bar

Dsl71, 2024-04-21:
cmd shell was still open:

[image]

Dsl71, 2024-04-21:
Boot
Let GC run some time
Under Instantaneous screen press menu
leave menu

Gissio, 2024-04-21:
Let me understand. Are you reporting two issues? The flashing problem, and the instantaneous view bar disappearing after you return from the menu?

Does the second issue appear if you flash with stlink.exe? Could you post a brief video?

Dsl71, 2024-04-21:
Ah i found with short press power you can change the bar... so no bug

Gissio, 2024-04-21:
I'm glad. Thanks for the feedback!

Gissio, 2024-04-21:
Could you edit `flash-fs2011-gd32f150c8.bat` and change the content to:

    @echo off
    flashtool.bat "radpro-fs2011-gd32f150c8" "-c ""set CPUTAPID 0"" -f target/stm32f1x.cfg -c ""init; reset halt; dump_image BACKUP_PATH 0x08000000 0x10000; exit;""" "-c ""set CPUTAPID 0"" -f target/stm32f1x.cfg -c ""init; reset halt; stm32f1x mass_erase 0; program IMAGE_PATH 0x08000000; reset run; exit;""" %1

Dsl71, 2024-04-22:
c:\temp\ft>flash-fs2011-gd32f150c8.bat
Backing up old firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
0
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.223576
Info : [stm32f1x.cpu] Cortex-M3 r2p1 processor detected
Info : [stm32f1x.cpu] target has 6 breakpoints, 4 watchpoints
Info : [stm32f1x.cpu] Examination succeed
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f1x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080005d0 msp: 0x20002000
Flashing new firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
0
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.222004
Info : [stm32f1x.cpu] Cortex-M3 r2p1 processor detected
Info : [stm32f1x.cpu] target has 6 breakpoints, 4 watchpoints
Info : [stm32f1x.cpu] Examination succeed
Info : starting gdb server for stm32f1x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f1x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080005d0 msp: 0x20002000
Info : device id = 0x13030410
Info : flash size = 64 KiB
[stm32f1x.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Error: timed out while waiting for target halted
[stm32f1x.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0x61000003 pc: 0x08000620 msp: 0x20001fe0
Error: error waiting for target flash write algorithm
Error: error writing to flash at address 0x08000000 at offset 0x00000000
embedded:startup.tcl:1689: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 1754
at file "embedded:startup.tcl", line 1689
Done.
Drücken Sie eine beliebige Taste . . .

nope

Gissio, 2024-04-22:
Ok. Can you change the content of `flash-fs2011-gd32f150c8.bat`:

    @echo off
    flashtool.bat "radpro-fs2011-gd32f150c8" "-c ""set CPUTAPID 0"" -f target/stm32f0x.cfg -c ""init; reset halt; dump_image BACKUP_PATH 0x08000000 0x10000; exit;""" "-d3 -c ""set CPUTAPID 0"" -f target/stm32f0x.cfg -c ""init; reset halt; stm32f0x mass_erase 0; program IMAGE_PATH 0x08000000; reset run; exit;""" %1

This will fix the problem, but will produce a lot of debugging output. Could you post it in a `.txt` file?

Dsl71, 2024-04-23:
[log.txt](https://github.com/Gissio/radpro/files/15071032/log.txt)

Bitte

Gissio, 2024-04-24:
It seems there was an SWD parity error (STLINK_SWD_AP_STICKY_ERROR), which should occur when there is a data transmission error.

Could you try again?

Dsl71, 2024-04-24:
ok, with the standard st-link tool it works like a charm

Gissio, 2025-01-14:
Can you try again?

[radpro-flashtool-2.1beta1.zip](https://github.com/user-attachments/files/18416058/radpro-flashtool-2.1beta1.zip)

DG0MG, 2025-02-09:
A few days ago I have updated an FS2011 to 2.0.3 without any issues.

DG0MG, 2025-02-12:
Yes, I used the batch file for Windows that is included in release 2.0.3. It worked without any problems.

DG0MG, 2025-02-12:
But I have the STM32F051 as a Processor. Perhabs this makes the difference?

Gissio, 2025-02-12:
DG0MG: My bad! I confused with with Dsl71!

Gissio, 2025-05-04:
Closing as there is no follow up.

closed 2025-05-04 by Gissio