Gissio/radpro #73: Firmware boot halted

PsychoBitchy, 2024-05-03:
Hello.
No matter what I try on my FINRSI CH32F
- downloading and uploading various "install" built firmware from repo.
- Building from source / signing, from the last repo version.

The device boots up just after the flash and is running fine, but if I switch the device off then on again, at some point the device stays stucked on "Radpro version" screen then blacks out. In that state, not even the power button works. It's in limbo state.

I have to reset then reflash to bring it back to life.

Any ideas?

[image]

PsychoBitchy, 2024-05-03:
**Edit: The device eventually goes back to life, but after a solid 30 seconds or so.
I had the impression it was stucked because at some point the boot process is much longer.**
Not a bug, closing

closed 2024-05-03 by PsychoBitchy

Gissio, 2024-05-06:
Most likely your CR1220 battery needs replacement!

I've added your questions in the [Q&A section of the user manual](https://github.com/Gissio/radpro/blob/main/docs/users.md#faq).

Gissio, 2024-05-07:
It's not actually hung, it's Rad Pro waiting for the microprocessor to start the real-time clock. The CH32 takes quite a while to do that.

Once you replace the battery, it should start right away.

PsychoBitchy, 2024-05-13:
Hello, just a small update on this case:
Replacing the board backup CR1220 battery indeed fixed every issues I had.
Thanks for pointing out and eagle-eye spotting the issue here.