Gissio/radpro #74: apm32f103rb GC-01 battery life 20 hours max, also crash on turn on / off during charging

Captainslarty, 2024-05-04:
Hello sir, as in the title. The GC-01 with apm32f103rb with latest firmware has a battery life of about 20 hours maximum (screen off!). Also, and it may hint to other issues ??? - turning on or off the unit whilst charging causes a blank screen needing a hard reset or reload of firmware.

I also note the charging voltage is far too high for the battery - reaching 4.65V ! . I am unaware if you have control over this.

Thank you sir. Joe

dc1rdb, 2024-05-05:
Hi Joe, I just checked your issues on my unit (GC-01, also with Geehy processor).
I could not reproduce any of them. Battery life on my unit is close to 40 hours (HV profile custom, 5kHz, 1.5%) and turning the unit on or off whilst charging does not have any negative effects either.
My charging voltage peaks at 4.2 Volts.
Maybe there is a hardware fault in your unit.
Regards, Robert

dc1rdb, 2024-05-07:
Current draw on my GC-01 with RadPro 2.0rc4 is around 29 mA with sleeping display, a significant improvement over the stock firmware. Custom HV profile makes quite a difference :)

Captainslarty, 2024-05-07:
Hello guys, sorry for the delay in replying but I have been away. Interesting findings. 
dc1rdb, I need to check the current consumption on mine when I get home. Regarding custom HV profiles ?. How are you measuring the HV ?. I intend to check mine and to that end have just received some 1Gohm (1000M!) 1% resistors to assist.
Gissio, thanks for the response. If you plug the unit in to usb when off, then, turn it on to check, say, the charge / battery voltage ?.. does it not immediately go into USB Drive mode ? - mine does - and needs a reset to return from that unless I load firmware which obviously I do not want to do ;-).  Out of interest, the original firmware you have will not load - the usb disconnects, yet the custom firmware will load every time so no issues with USB. There is no way for me to restore an original firmware though and of course, the need to reset if you turn it on after it has been connected as it enters firmware update mode. Can you also explain to me how the device is measuring the battery volage ? - I am unsure as I though it would be a tap from the battery itself to an adc but I cannot see one apart from RA0 but surely this is also measuring part of the usb input via the diode across the 4054 when plugged in ? (LIke you, it would seem I also do not see what the mosfet is doing after the battery !! :)) ... many thanks. Joe

Captainslarty, 2024-05-07:
Hi Gissio - as for studio326's findings, I am as baffled as you ! :-)
Joe

dc1rdb, 2024-05-07:
Hi Joe, I am measuring the voltage across the geiger tube with a Fluke 80K-40 HV probe.

Btw, I was able to load the original firmware via ST-Link, even though I screwed up my original bootloader so it won't go int USB drive mode at all anymore :)

dc1rdb, 2024-05-07:
Joe, would you be willing to share your bootloader as per Gissio's instructions?
https://github.com/Gissio/radpro/discussions/59
Apparently, there are a number of different versions out there in the wild.
Thanks in advance, Robert

Captainslarty, 2024-05-07:
Hi Robert, Thanks for the info. Seems like the Fluke HV Probe is probably also 1G ohm impedance !. so results should be similar with a 10M ohm input meter.
I would be happy to share my bootloader but I have the different processor - the apm32f103rb. I do not know if those instructions would work ?
Regards. Joe

dc1rdb, 2024-05-07:
I'm not 100% sure if the instructions apply. Let's see what Gisdio thinks :)

Captainslarty, 2024-05-07:
Sure my friend, I would think not though as the processor is quite a bit different.It would seem the same applies here though with the apm - different versions - so maybe another version of the tool ? - but only if it is worth the effort of course. I do have a few st-link in the bits box lol :-).Regards,Joe

Gissio, 2024-05-07:
It actually works also for the CH32. It would be great if you could do that.

Captainslarty, 2024-05-08:
[Joes Blue GC-01 Bootloader.zip](https://github.com/Gissio/radpro/files/15244278/Joes.Blue.GC-01.Bootloader.zip)

Here's the bootloader dump Gissio my friend. I will upload images of the unit later today to match the bootloader title.
Tell me if it checks out ok or if I need to do anything else to assist ?
Joe

Captainslarty, 2024-05-08:
Oh sorry, just to add, It is actually a text file - I should have given it a txt suffix - apologies - but it will load into an editor ok - I used notepad++
Joe

Captainslarty, 2024-05-08:
Hello Gissio and friends, here are the images for MY GC-01. (Joe's Blue GC01)
Notice - the ONLY FNISRI marking is the label inside - the advert, however - cleary says FNISRI https://pt.banggood.com/FNIRSI-GC-01-Geiger-Counter-Nuclear-Radiation-Detector-Personal-Dosimeter-X-ray--ray--ray-Radioactivity-Tester-Marble-Detector-p-1996587.html?rmmds=myorder&cur_warehouse=CN

However, it also states it has the large tube, not the smaller tube ! - hence I obtained a refund of 18.29 Euro from bangood for the price of a 4011 tube from aliexpress. I show the new tube to be fitted later in one of the following images which I hope are helpful.

let me know if you need more details / images etc etc.
Joe

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Gissio, 2024-05-08:
Thanks for the bootloader. Would you be so kind to cross-post in #59, so things stay in one place? Thanks.

Captainslarty, 2024-05-09:
Posted to #59 as requested - along with images.

Question my friend ? - I have a new unit on order from FNISRI Store and wonder if you have a way to dump the original FIRMWARE as well ?
Joe

Captainslarty, 2024-05-09:
All sorted

Captainslarty, 2024-05-13:
Hello gissio.
Question my friend ? - I have a new genuine unit on order from FNISRI Store and wonder if you have a way to dump the original FIRMWARE prior to ANY modifications ? . Please can you advise ?
Joe

Gissio, 2024-05-14:
Let me point you to #24.

It is possible to dump the bootloader, but not the original firmware.

Captainslarty, 2024-05-14:
Thank you Gissio

Captainslarty, 2024-05-15:
Robert, to reload the original firmware use the fnirsi-gc01-ch32f103r8t6-v1.5.bin !!
Yes, it's for the apparently 'wrong' processor but it works fine. (that is IF you can enter the boot loader.)

Gissio sir - the APM ORIGINAL fnirsi-gc01-apm32f103rbt6-v1.6.bin firmware does not work here yet the above fnirsi-gc01-ch32f103r8t6-v1.5.bin does ? - can you explain please ? - definitely marked as APM processor.

Joe

dc1rdb, 2024-05-15:
Thanks for the info, Joe.
Currently, I can not enter the bootloader, but it should also be loadable via ST-Link. I might give it a try.

Captainslarty, 2024-05-16:
I Robert, It is not perfect in some of the settings, but in main use it is absolutely fine. I simply cannot get ver 106 to at all under any circumstances.

I have also emailed Fnirsi for a suitable firmware - fingers crossed... be useful for comparisons. Best wishes my friend. Joe

dc1rdb, 2024-05-16:
Hi Joe, I would load the original firmware only temporarily anyway. Gissio's RadPro is so much better and turns this cheap Geiger counter into a serious instrument.
Best wishes, Robert

Captainslarty, 2024-05-17:
Hi Robert, that was the only intention so as to check the current draw / battery life. There is no way it is staying on original firmware. Gissio firmware is the icing in the cake. 🥇 
Joe

Captainslarty, 2024-05-27:
Hello guys. Here is how to check the HV voltage on a GC-01 using a standard 10Mohm input meter. Image and details are direct from the manufacturer.
Should be 380 - 395, notice the meter lead orientation and placement. My test one is 385V - standard firmware new meter, un modified. Definitely worth noting. Sorry about low quality, that was what they provided in AliX chat. (I had a new meter from them with faulty tube so they are sending me another, and this was to assist with fault finding. REALLY useful info as I was originally measuring from battery ground to HV.

[image]

Captainslarty, 2024-05-27:
Also good info jor all normally used tubes (and more)
[image]

Captainslarty, 2024-05-29:
For Gissio and Robert.

Here is the ACTUAL GENUINE FACTORY firmware - the latest - for the GC-01. It works to revert my 'small tube' version back to factory so I can now load either firmware and then reload the original to test voltages etc. 

Gissio my friend, please add it to the repository for the GC-01 with APM processor. Works perfectly, v 1.6 does not on mine.

[GC01_V1.6-2.zip](https://github.com/Gissio/radpro/files/15479874/GC01_V1.6-2.zip)

Cheers guys
Joe

Send virtual beers please lol.... :-)

hackra76, 2024-05-29:
> Hello guys. Here is how to check the HV voltage on a GC-01 using a standard 10Mohm input meter. Image and details are direct from the manufacturer. Should be 380 - 395, (with the proper full size tube - J321 here - 4011 wtx is more or less the same for all practical purposes). notice the meter lead orientation and placement. My test one is 385V - standard firmware new meter, un modified. Definitely worth noting. Sorry about low quality, that was what they provided in AliX chat. (I had a new meter from them with faulty tube so they are sending me another, and this was to assist with fault finding. REALLY useful info as I was originally measuring from battery ground to HV.
> 
> ![factory hv measuring point should be 385 to 395 vdc](https://private-user-images.githubusercontent.com/34556125/333979570-c991257a-9da6-4046-8889-efd01a64a64e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5NzUwMDMsIm5iZiI6MTcxNjk3NDcwMywicGF0aCI6Ii8zNDU1NjEyNS8zMzM5Nzk1NzAtYzk5MTI1N2EtOWRhNi00MDQ2LTg4ODktZWZkMDFhNjRhNjRlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI5VDA5MjUwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjYjdiN2NkYTU3NzYwMGZmNTZiZjI4MTU1OWY5MDRlYThjMTBmZDNlNDliNjc0MDBiNzQzNzViYzlkODU2OGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.z_ooUp4nJBn_Y8SK6SMc7XdsfeDp-Tnj5_7YWJOLLRM)

Hi, I tried measuring the voltage across the points as in your image and no matter what tube type I set, the PWM frequency or Duty cycle I set, I always measure 88-90 V..... Very strange. My unit is a APM version with J321 swapped tube. Strange thing is, the device seems to measure more or less correctly, the background is consistent with my BR-9C unit, and it registers the smoke detector source also...
[image]

Captainslarty, 2024-05-29:
what is the input impedance of your multimeter ?

 
 

20240529_112325.2.jpg (view on web)

hackra76, 2024-05-29:
According to reviewit is 10-11 MOhm.....

https://lygte-info.dk/review/DMMAnengQ1%20UK.html

Captainslarty, 2024-05-29:
then there is something wrong....... the data is from the factory and works great on 3 gc-01 units.

 
 

https://lygte-info.dk/review/DMMAnengQ1%20UK.html

Captainslarty, 2024-05-30:
Hi Gissio, I do not see that here. The file size differences are absolutely huge to start with.Your version semi works but the text on the menus are missing. The version I posted from Fnirsi works superbly. Please can you advise how the file sizes can be so dramatically different ?
ThanksJoe

 
 

The file you submitted is identical to https://github.com/Gissio/radpro/blob/main/docs/devices/FNIRSI%20GC-01/firmware/fnirsi-gc01-apm32f103rbt6-v1.6.bin

Captainslarty, 2024-05-30:
Hmm, maybe it is something to do with how the file is being downloaded from github ? - it was around 423kB, then I tried 'raw' and it came down to sensible size. Yet, the large file still loaded ? 

Maybe I am doing something wrong with the download from github ?

Gissio, 2024-05-30:
Try downloading the file you uploaded. You will notice it is the same as the file I pointed you to.

Captainslarty, 2024-05-31:
Thanks. from the list when searching for your files on github, I was right clicking the file and downloading it that way, this gave a huge file size, still a bin file and still loaded but failed to work - understandable now ;) .
I had not seen the download raw option.
Git hub is not really for user friendly unless you are a coder - in my opinion of course.
I was looking at the Bosean 5000 version you did - for example - and cannot locate the file from the thread on the radpro version for it. The links seem to end at an option with no firmware. I can find the loader, but not the file to load. I know this is off topic, but I am using it as an example of how difficult it can be to navigate. I thought I was computer literate but github proves me wrong lol.
Joe

Gissio, 2024-06-07:
Closing issue due to inactivity.

closed 2024-06-07 by Gissio