Gissio/radpro #81: FNIRSI GC-01_V0.2 can not flash radpro and unable to restore to factory defaults

mveplus, 2024-05-25:
FNIRSI GC-01_V0.2 with a CH32F103R8T6 (WCH) processor. Unable to flash with rc4 or rc5.
Initially the device was showing up as mass storage, but was not reacting after successfully copy of the binary. I did a few sequential test and variations:
* Connecting to USB - starting form off, momentary pressing `ok/power` device shows as mass storage `GC01 BOOT`,  with single empty Readme.txt, dragging and dropping `radpro-fnirsi-gc01-ch32f103r8-2.0rc4.bin` file copied successfully but device did not reboot. 
* Connecting to USB - starting form off, momentary pressing and holding `ok/power` device shows as mass storage `GC01 BOOT`,  with single empty Readme.txt, dragging and dropping `radpro-fnirsi-gc01-ch32f103r8-2.0rc4.bin` file copied successfully but device did not reboot. After releasing button mass storage disconnects but the firmware was not flashed. 
*  Connecting to USB - starting form off, momentary pressing `ok/power` device shows as mass storage `GC01 BOOT`,  with single empty Readme.txt, holding right arrow[cogwheel] and  key dragging and dropping `radpro-fnirsi-gc01-ch32f103r8-2.0rc4.bin` file copied successfully but device did not reboot, releasing the arrow there was no change.
* tested the same senescence with [`fnirsi-gc01-ch32f103r8t6-v0.3.bin` and `fnirsi-gc01-ch32f103r8t6-v1.5.bin`](https://github.com/Gissio/radpro/tree/main/docs/devices/FNIRSI%20GC-01/firmware) no response. 

At that stage the original firmware was still fully functional. 

Then I moved to the [alternative flash instructions](https://github.com/Gissio/radpro/blob/main/docs/devices/FNIRSI%20GC-01/install-stlink.md). As I did no have ST-LINK I used my J-LINK that supports SWD. Set-up and the JLink software recognised the chip. 
I flashed the `radpro-fnirsi-gc01-ch32f103r8-2.0rc4.bin` it did flashed successfully but after that the device was not starting anymore, or showing as mass-storage device. 

Unfortunately I did not save  the original, before overriting it, not the only thing that still works is the bootloader, and the mass-storage is showing up as a drive, but cant power up. 

I have tied flashing all binaries and test with each one, but no change from the above. Example log output from the J-LInk:

```
Opening data file [~/Downloads/fnirsi-gc01-ch32f103r8t6-bootloader.bin] ...
 - Data file opened successfully (16384 bytes, 1 range, CRC of data = 0x84FA8749, CRC of file = 0x84FA8749)
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link ARM V8 compiled Nov 28 2014 13:44:46
 - Probe/ Programmer S/N: 788529158
 - Device "CS32F103C8" selected.
 - Target interface speed: 4000 kHz (Fixed)
 - VTarget = 3.319V
 - InitTarget() start
 - SWD selected. Executing JTAG -> SWD switching sequence.
 - DAP initialized successfully.
 - InitTarget() end - Took 19.0ms
 - Found SW-DP with ID 0x2BA01477
 - DPv0 detected
 - CoreSight SoC-400 or earlier
 - Scanning AP map to find all available APs
 - AP[1]: Stopped AP scan as end of AP map has been reached
 - AP[0]: AHB-AP (IDR: 0x24770011)
 - Iterating through AP map to find AHB-AP to use
 - AP[0]: Core found
 - AP[0]: AHB-AP ROM base: 0xE00FF000
 - CPUID register: 0x412FC231. Implementer code: 0x41 (ARM)
 - Found Cortex-M3 r2p1, Little endian.
 - FPUnit: 6 code (BP) slots and 2 literal slots
 - CoreSight components:
 - ROMTbl[0] @ E00FF000
 - [0][0]: E000E000 CID B105E00D PID 000BB000 SCS
 - [0][1]: E0001000 CID B105E00D PID 003BB002 DWT
 - [0][2]: E0002000 CID B105E00D PID 002BB003 FPB
 - [0][3]: E0000000 CID B105E00D PID 003BB001 ITM
 - [0][4]: E0040000 CID B105900D PID 003BB923 TPIU-Lite
 - Executing init sequence ...
  - Initialized successfully
 - Target interface speed: 4000 kHz (Fixed)
 - Found 1 JTAG device. Core ID: 0x2BA01477 (None)
 - Connected successfully
Auto programming target (16384 bytes, 1 range) ...
 - Checking if selected data fits into selected flash sectors.
 - Start of preparing flash programming
 - End of preparing flash programming
 - Start of determining dirty areas in flash cache
 - End of determining dirty areas
 - CPU speed could not be measured.
 - Start of erasing sectors
 - Blank checking 0x08000000 - 0x08003FFF
 - Erasing range 0x08000000 - 0x08003FFF ( 16 Sectors, 16 KB)
 - End of erasing sectors
 - Start of flash programming
 - Programming range 0x08000000 - 0x08003FFF ( 16 Sectors, 16 KB)
 - End of flash programming
 - Flash programming performed for 1 range (16384 bytes)
 - 0x8000000 - 0x8003FFF ( 16 Sectors, 16 KB)
 - Start of verifying flash
 - End of verifying flash
 - Start of restoring
 - End of restoring
 - Executing exit sequence ...
  - De-initialized successfully
 - Target erased, programmed and verified successfully - Completed after 23.415 sec
``` 
  
Any advice or help would be appreciated, I do not want to run the original firmware but radpro one instead, mainly as I want to be able to read the data form the sensor via USB. Some [photos](https://www.icloud.com/sharedalbum/#B0y5BydzF9y5DB)

mveplus, 2024-05-25:
Okay, you can disregard all the above. 
I manage to overwrite the boot-loader and did not flash the binary starting from the correct address boundaries for the program data. All fixed now ;) I can flash both the original and the radpro with J-Link. 
Looking at the flashtoo.sh I realized I have two binaries to flash ;P 
bootloader.bin 0x08000000 and the radpro.bin[program]  0x08004000
I also found my original binary provided by to the manufacturer, it has the same md5 hash as the one of the origiaal recover versions in the repo:
* 54efd3ec9995e948db7c8ddae235a2c7  GC01_V0.3.bin
* 54efd3ec9995e948db7c8ddae235a2c7  fnirsi-gc01-ch32f103r8t6-v0.3.bin

Thanks for providieng all the files, and specially the bootloader binary ;)

closed 2024-05-25 by mveplus

Gissio, 2024-05-25:
I'm glad you manage to sort things out!