Gissio/radpro #83: FNIRSI GC-01: Serial Link comunication

mveplus, 2024-05-25:
Installed Rad Pro on GC-01's with a CH32F103R8T6 processor (WCH) `radpro-fnirsi-gc01-ch32f103r8-2.0rc4.bin`.
All work beautifully on the device, thank you!  I would like to use the serial interface to read the data for the device but have some issues on Linux enumeration the port correctly. 
```
May 25 22:12:49 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:12:49 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:12:49 b54583.lan kernel: usb 3-5.3.2.3: device not accepting address 88, error -71
May 25 22:12:49 b54583.lan kernel: usb 3-5.3.2.3: WARN: invalid context state for evaluate context command.
May 25 22:12:49 b54583.lan kernel: usb 3-5.3.2-port3: unable to enumerate USB device
May 25 22:12:49 b54583.lan kernel: usb 3-10: reset full-speed USB device number 3 using xhci_hcd
May 25 22:12:49 b54583.lan kernel: usb 3-10: reset full-speed USB device number 3 using xhci_hcd
May 25 22:21:49 b54583.lan kernel: perf: interrupt took too long (2508 > 2500), lowering kernel.perf_event_max_sample_rate to 79000
May 25 22:22:20 b54583.lan kernel: usb 3-10: reset full-speed USB device number 3 using xhci_hcd
May 25 22:22:20 b54583.lan kernel: usb 3-10: reset full-speed USB device number 3 using xhci_hcd
May 25 22:22:31 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 89 using xhci_hcd
May 25 22:22:31 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:22:31 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:22:32 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 90 using xhci_hcd
May 25 22:22:32 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:22:32 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:22:32 b54583.lan kernel: usb 3-5.3.2-port3: attempt power cycle
May 25 22:22:34 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 91 using xhci_hcd
May 25 22:22:34 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:22:34 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:22:34 b54583.lan kernel: usb 3-5.3.2.3: device not accepting address 91, error -71
May 25 22:22:34 b54583.lan kernel: usb 3-5.3.2.3: WARN: invalid context state for evaluate context command.
May 25 22:22:34 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 92 using xhci_hcd
May 25 22:22:34 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:22:35 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:22:35 b54583.lan kernel: usb 3-5.3.2.3: device not accepting address 92, error -71
May 25 22:22:35 b54583.lan kernel: usb 3-5.3.2.3: WARN: invalid context state for evaluate context command.
May 25 22:22:35 b54583.lan kernel: usb 3-5.3.2-port3: unable to enumerate USB device
May 25 22:24:17 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 93 using xhci_hcd
May 25 22:24:17 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:24:17 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:24:17 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 94 using xhci_hcd
May 25 22:24:17 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:24:18 b54583.lan kernel: usb 3-5.3.2.3: device descriptor read/64, error -32
May 25 22:24:18 b54583.lan kernel: usb 3-5.3.2-port3: attempt power cycle
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 95 using xhci_hcd
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: device not accepting address 95, error -71
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: WARN: invalid context state for evaluate context command.
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 96 using xhci_hcd
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: Device not responding to setup address.
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: device not accepting address 96, error -71
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2.3: WARN: invalid context state for evaluate context command.
May 25 22:24:20 b54583.lan kernel: usb 3-5.3.2-port3: unable to enumerate USB device
May 25 22:30:03 b54583.lan kernel: usb 3-10: reset full-speed USB device number 3 using xhci_hcd
May 25 22:30:03 b54583.lan kernel: usb 3-10: reset full-speed USB device number 3 using xhci_hcd
```

Mass storage is enumerated correctly: 
```
May 25 22:58:07 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 97 using xhci_hcd
May 25 22:58:07 b54583.lan kernel: usb 3-5.3.2.3: New USB device found, idVendor=0483, idProduct=002a, bcdDevice= 2.00
May 25 22:58:07 b54583.lan kernel: usb 3-5.3.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 25 22:58:07 b54583.lan kernel: usb 3-5.3.2.3: Product: AIR01
May 25 22:58:07 b54583.lan kernel: usb 3-5.3.2.3: Manufacturer: FNIRSI
May 25 22:58:07 b54583.lan kernel: [44B blob data]
May 25 22:58:07 b54583.lan kernel: usb-storage 3-5.3.2.3:1.0: USB Mass Storage device detected
May 25 22:58:07 b54583.lan kernel: scsi host1: usb-storage 3-5.3.2.3:1.0
May 25 22:58:08 b54583.lan kernel: scsi 1:0:0:0: Direct-Access     AIR      DETECTION 1.0         PQ: 0 ANSI: 2
May 25 22:58:08 b54583.lan kernel: sd 1:0:0:0: Attached scsi generic sg0 type 0
May 25 22:58:08 b54583.lan kernel: sd 1:0:0:0: [sdb] 51200 2048-byte logical blocks: (105 MB/100 MiB)
May 25 22:58:08 b54583.lan kernel: sd 1:0:0:0: [sdb] Write Protect is off
May 25 22:58:08 b54583.lan kernel: sd 1:0:0:0: [sdb] Mode Sense: 03 00 00 00
May 25 22:58:08 b54583.lan kernel: sd 1:0:0:0: [sdb] No Caching mode page found
May 25 22:58:08 b54583.lan kernel: sd 1:0:0:0: [sdb] Assuming drive cache: write through
May 25 22:58:08 b54583.lan kernel: sd 1:0:0:0: [sdb] Attached SCSI removable disk
``` 
Is USB-C to serial bridge supported on GC-01 with WCH MCP?

Gissio, 2024-05-25:
Please read the manual: https://github.com/Gissio/radpro/blob/main/docs/devices/FNIRSI%20GC-01/install.md#fnirsi-gc-01-with-a-ch32f103r8-wch-microprocessor

closed 2024-05-25 by Gissio

mveplus, 2024-05-25:
Thank you @Gissio I was looking for it under `Data logging` did not realise the settings menu have much more options [ a minor UI improvement with line indicator on a side were you are ;)  would be nice to have  ]

The port comes up, but it's a bit hit and miss, it's sometimes does, and other times you have to press more than once on the `data mode` to activate it. The enable state is not remembered if device is powered-off [ I guess battery power saving reasons ? ] it would be nice if it can be set `on` and `state` remembered with warning that will impact if only on battery life or turn off if USB power is not connected.  

```
May 25 23:14:29 b54583.lan kernel: usb 3-5.3.2.3: new full-speed USB device number 112 using xhci_hcd
May 25 23:14:35 b54583.lan kernel: usb 3-5.3.2.3: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 1.00
May 25 23:14:35 b54583.lan kernel: usb 3-5.3.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=254
May 25 23:14:35 b54583.lan kernel: usb 3-5.3.2.3: Product: Rad Pro
May 25 23:14:35 b54583.lan kernel: usb 3-5.3.2.3: Manufacturer: Gissio
May 25 23:14:35 b54583.lan kernel: usb 3-5.3.2.3: SerialNumber: 0CDE7343
May 25 23:14:35 b54583.lan kernel: cdc_acm 3-5.3.2.3:1.0: ttyACM0: USB ACM device
````   

Now, I can connect to it but device does not seem respond to any of the examples. [ using baud-rate 115200 and settings as advised] , any advice?

Gissio, 2024-05-26:
It's a great idea to store the data mode state in the settings.

If you have trouble connecting, check the USB cable.

To test, try [https://github.com/Gissio/geigerlog-radpro](https://github.com/Gissio/geigerlog-radpro).

mveplus, 2024-05-26:
Good work! looks great ;) , Can the `geigerlog-radpro` work without the GUI only in CLI? I had a quick look but can't see an option to start without the GUI. 

```
==== Rad Pro Device Info =============================================================
Configured Connection: Port:"/dev/ttyACM0"
Device connected.             
Hardware ID                     : FNIRSI GC-01 (CH32F103C8)
Software ID                     : Rad Pro 2.0rc4
Device ID                       : 89eb5688
Device battery voltage          : 4.866V
Device time                     : 2024-05-26 09:57:31
Tube life time                  : 12:47:40
Tube life pulse count           : 16762
Tube rate                       : 37.218 CPM
Tube conversion factor          : 153.800 CPM/µSv/h
Tube dead time                  : 0.0001540 s
Tube dead-time compensation     : 0.0000000 s
Tube background compensation    : 0.000 CPM
Tube HV PWM frequency           : 9207.16 Hz
Tube HV PWM duty cycle          : 0.7500 %
```

Okay I think I found out more about "enable `data mode` sometimes the `USB ACM device` does not show up". 
* first time I found the option and enable it was enumerated immediately in my OS, so I was expecting this to happen every time. 
* then I turned off the device and then back and connected the USB link before enabling it, then when I did enable `data mode` nothing was happening in the `dmseg` logs, I had to repeard going to the menu and re-enable and it sometimes it shows up, but other times it will not sense any change on the OS side with the rad-pro link - timing issues? 
* Then while the `data mode` was enabled [ staying on the `Data connection enabled ` screen ], Unplug and re-plug the USB link worked every time enumerating it. 
* Also tested with diferent cable to rule out `bad cable`. 

Thanks for the heads up for the `state` .  Let me know if I can help with the device I have? Cheers

Gissio, 2024-05-27:
For a CLI interface, please read the manual: [https://github.com/Gissio/radpro/blob/main/docs/users.md#radpro-tool](https://github.com/Gissio/radpro/blob/main/docs/users.md#radpro-tool)