Gissio/radpro #87: 2.0rc5 on FNIRSI GC-01

ullix, 2024-05-29:
Installation of rc5 was straight forward, and code is running on GeigerLog.
Most interesting to me is the feature allowing PWM setting. But has anything been changed there? It seems I am getting different voltages - much higher - than with the same settings on rc3?

Gissio, 2024-05-29:
What do other users see?

dc1rdb, 2024-05-29:
GC-01 with Geehy processor here and setting HV via setup menu.
With 2.0rc4, 5kHz @ 1.5% resulted in 440V.
With 2.0rc5, the same setting produces close to 500V. I had to change to 5kHz @ 1% to get back to 440V.

Caution: the profile "energy saving alt" results in voltages over 900V!

alfmck, 2024-05-29:
I didn't have the change. My GC-01 with CH32F103R8T6. Settings 10kHz&2.5% 440V. 2.0rc5 I built from src, supplied in radpro releases.

Gissio, 2024-05-29:
To fix this, I would need to know:

* Exactly what device you are using (including MCU).
* What HV profile/custom HV profile you are using.
* Confirmation that on version X it worked as expected, and now it doesn't. It would help greatly if you could measure the actual PWM value.

I just tested the code, and found it works as expected, without differences to the previous releases.

Gissio, 2024-05-30:
> GC-01 with Geehy processor here and setting HV via setup menu. With 2.0rc4, 5kHz @ 1.5% resulted in 440V. With 2.0rc5, the same setting produces close to 500V. I had to change to 5kHz @ 1% to get back to 440V.
> 
> Caution: the profile "energy saving alt" results in voltages over 900V!

Are you setting the custom profile through the menus or data communications?

dc1rdb, 2024-05-30:
I'm not using data communication. Setting everything via the setup menu.

ullix, 2024-05-30:
I have posted an analysis of PWM setting and Anode Voltage here: https://www.geigerzaehlerforum.de/index.php?msg=30457 All settings done programmatically (and read back to verify).

While the post is in German, I believe the graphs can be understood by themselves.

One question: why is freq and duty set in fixed settings only? AFAK the STM32s allow floating points for both?

TzOk83, 2024-05-30:
For my GC-01 unit with J614(?) tube on factory default setting I have ~750V, and on energy-saving ~450V. I'm measuring using a 100MOhm input impedance DC volt meter with a 1000:1 HV probe.

Gissio, 2024-05-30:
TzOk83: that is way too high.

Can somebody measure the frequency and duty cycle at the left of the R9 resistor, and confirm it matches the menu settings? R9 is below the LCD ribbon cable.

Gissio, 2024-05-30:
Rad Pro limits options to the ones allowed in the user interface, in order to apply the KISS principle.

TzOk83, 2024-05-30:
These are close to real voltage measurements, as my effective input impedance is about 10 GOhm. So I'm putting virtually no load on the voltage source.

The measured frequency is exactly 5kHz, and the duty cycle is about 1.3% (th=2.6us).

Gissio, 2024-05-30:
Thank you so much. So frequency and duty cycle seem to be working correctly. It would be helpful if this measurement could be repeated with rc4!

Gissio, 2024-05-30:
TzOk83:

Sorry, I did not mean to imply that your measurements are wrong! Quite the contrary. I was just noticing that Rad Pro's default profile should not produce 750 V!

And thank you so much for measuring frequency and duty cycle. They seem to be working correctly. It would be helpful if this measurement could be repeated with rc4, to check for any difference.

TzOk83, 2024-05-30:
I've done measurements on rc3, rc4, and rc5. They are all the same:
Factory default: 740V, 46.06kHz, 50%
Energy-saving: 460V, 5kHz, 1.5%

Gissio, 2024-05-30:
TzOk83: excellent, thanks for your work!

How come some users report changes in the high voltage?

TzOk83, 2024-05-30:
Measuring this voltage is quite tricky, because of the extremely low current capabilities of the source. Even with my gear, I can notice that the voltage gradually drops when I connect the probe, yet it doesn't make a difference whether I measure on the indicated capacitor, or directly on the GM tube terminals.
Another thing is that this voltage will vary depending on the battery voltage.

Gissio, 2024-05-30:
What you say makes a lot of sense. Thanks a lot!

closed 2024-05-30 by Gissio

ullix, 2024-05-31:
From @TzOk83 
> I've done measurements on rc3, rc4, and rc5. They are all the same:
> Factory default: 740V, 47.06kHz, 50%

This surely raises new questions: The rc5 software does not allow the exact same settings, but it is possible to set 40kHz and 50% via the `SET tubeHVDutyCycle` and `SET tubeHVFrequency` commands. 

When doing this setting I get a voltage of 305V (via 1GOhm to DMM); less than half of factory default. This is not even enough to reach the Geiger plateau, and so there are no counts whatsoever. 

I can do a full scan of all Duty Cycle values from 1% to 90%(firmware limit), and get this result:
[image]

The voltage barely exceeds 300V, and thus counts never occur, except for this peculiar spike (reproducible!) after which voltage drops to near zero.

I believe freq and duty are the only parameters relevant for the voltage, but if so, how do you reconcile this significant difference between factory and Radpro firmware?

TzOk83, 2024-05-31:
I was not using PC software but was using the inbuilt menu.

GM PSU in GC-01 works with open loop, so output voltage depends only on: frequency, duty cycle, and **battery voltage**.

What is the input impedance of your DMM? Between which points are you measuring anode voltage?

ullix, 2024-05-31:
As I am measuring from a PC the GC-01 obviously has to be connected via USB, so batt voltage should always be at max; the firmware says under Statistics: "DC voltage  4.753 V".

The DMM has impedance specs ">10M"; when measured from a 2nd DMM I get 11.05M.

I am measuring HV at the tube's anode and cathode end caps.

TzOk83, 2024-05-31:
I was measuring between tube "-" and C12 "+", but it is roughly 10V difference between C12 "+" and GM tube "+". I have disassembled my HV probe and it is a resistive voltage divider with 2 (I was sure, there was only 1, and the second one is the meter itself): 1000 Meg, and 1Meg, so it is 1 Gig Ohm input impedance, a way lower than I initially thought.

My unit was running on an internal battery, not connected to the mains. Remember that GM Tube "-" terminal is not at the same potential as USB GND! You can't short them. As I had an oscilloscope connected for frequency and duty cycle measurement, I couldn't have the USB plugged in.

ullix, 2024-05-31:
So, your setup is very similar to mine, I have 1G from anode to DMM input. I am aware of the GND and minus differences, but my DMM is run on battery, so this isn't impacted by it.

I just wondered whether this odd spike is due to increasing voltage, so I also programmed the reverse. Duty (black, right scale) goes from 1% to 90%, and down to 1 again. Green (right scale) is the anode voltage. Irritating symmetry :-/

[image]

TzOk83, 2024-05-31:
It looks like something in your unit is breaking down/starting to massively leak current above 300V.

ullix, 2024-06-01:
I don't think so; 1200V can be reached (which I find quite impressive for this device) 

[image]

but not for 40kHz. When I keep the duty cycle fixed, and scan through the frequencies, I get the result shown next.

Very clearly, the voltage drops with increasing frequency. No way in hell that at 47.06kHz @50% (what you showed) I could get anywhere near 740V. Unless there is some magic resonance involved, which, however, I can't test as the current firmware permits only preset values, and their max is 40kHz.

[image]

TzOk83, 2024-06-01:
I was not experimenting with 40kHz, only 5kHz with a variable duty cycle, and "factory default" setting, which produced 47kHz and 50% duty cycle.

ullix, 2024-06-01:
In my experience the counter with rc5 did not accept PWM commands unless the profile "Custom profile" was explicitly checked. It would not return an error, but just kept the defaults.

ullix, 2024-06-01:
Want to illuminate your desk with a Geiger tube? Here you go:
https://www.youtube.com/watch?v=avdHJTEzvmQ

Has the tube been damaged by too high voltage? In the video for some time the tube had been at normal voltage 400...500V, yet it was still lighting up. I had noticed the tube is quite light sensitive.

ullix, 2024-06-07:
The Rad Pro 2.0 final release has removed the restrictions for PWM settings when done from a PC; very nice.

So I could set the factory default conditions, which @TzOk83 had reported. 
> ... Factory default: 740V, 47.06kHz, 50%

[image]

But instead of also getting an anode voltage of 740V, I am getting much, much less, namely only 324V. This is even below the minimum voltage the tube requires to fire any counts, and indeed, the tube isn't counting anything. Not even with a source which should have produced some CPS=20 counts. 

So how could this possibly have been the factory default which would have rendered the counter incapable to produce any counts?

Freq and Duty are as reported back from the Radpro firmware. The voltage is measured by DMM after a 1G resistor to the anode.

What is going on?

Gissio, 2024-06-18:
> [TzOk83](/TzOk83)

Would you be so kind to test #96? I am unable to reproduce any of the errors the user reports.