Gissio/radpro #88: Bosean FS-5000 cannot flash with STLink

ricsim, 2024-05-29:
I have 3 devices.
Fnirsi GC-1 with WCH, that i just updated to 2.0rc5 with STLink.
Another GC-1 that I updated with the bootloader drive.
And a Bosean FS-5000 where I get the error below.
(Pinout is different than GC-1, but I rearranged the pins.)
Any ideas?

Backing up old firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J39S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.229005
Error: init mode failed (unable to connect to the target)

Flashing new firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J39S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.238497
Error: init mode failed (unable to connect to the target)

ricsim, 2024-05-29:
Here is a picture, on the picture one of the pins slipped out, but it was connecte when I tried to flash:
https://menedetter.net/bosean.png

tried it again and made sure all 4 pins are connected:
Backing up old firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J39S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.238497
Error: init mode failed (unable to connect to the target)

Flashing new firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J39S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.236915
Error: [stm32l4x.cpu] Examination failed
Warn : target stm32l4x.cpu examination failed
Info : starting gdb server for stm32l4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz

Done.

Gissio, 2024-05-29:
What happens if you keep the ok/power button pressed while flashing?

ricsim, 2024-05-29:
Many thanx!!
That worked, I am running RadPro now!!
Yeah ;)

Gissio, 2024-05-29:
Will update the docs.

closed 2024-05-29 by Gissio

ricsim, 2024-05-29:
I do not have a serial port over USB.
Windows shows no USB device, and also Linux does not show anything in dmesg, when i plug in the USB cable.
GC-01 with the same cable works fine.

Gissio, 2024-05-29:
Please open another issue.

Also, make sure you copy the backup in the `backup` folder to a safe place.

ricsim, 2024-05-30:
Strange, it seems to have been a one off.
Today it works perfectly! Thank you so much for the FS5000 support.

Here is the backup: https://menedetter.net/bosean-fs5000.bin