Gissio/radpro #89: Conversion factor for J613 GM tube

TzOk83, 2024-05-30:
I have noticed very low background radiation levels with this tube. In your description, I've found that you are using 64.8 cpm/µSv/h (9,5 cpm/mR/h) for both J613 and J614, but on the J614 datasheet I found online, they stated 5 cpm/mR/h, which is ~34,2 cpm/µSv/h. So I've set 33,9 cpm/µSv/h, and now my long-time average readings are more or less on par, with my old RKSB-104.

Gissio, 2024-05-30:
Could you please share the link to the J614 datasheet?

Supposing the HH614 is similiar to the J614 (they also look alike), according to [https://github.com/Gissio/radpro/blob/main/docs/tubes/HH614.jpg](https://github.com/Gissio/radpro/blob/main/docs/tubes/HH614.jpg), the conversion factor should be 10 cps/mR/h.

Also, it is a bad idea to compare background levels of different tubes, as the tubes themselves emit radiation. It would be more appropriate to compare the radiation levels with a test source.

TzOk83, 2024-05-30:
The one from Rad Pro is the one I've also seen, it is sometimes labeled G-M World. Nevertheless, when I enter 33.9 cpm/µSv/h I have readings ~0.1 uSv/h, which correspond to hourly updated National Atomic Agency reports for background radiation in my area (0.091 uSv/h). My RKSB-104 gives readings between 0.09 and 0.120 uSv/h.

Gissio, 2024-05-30:
Excellent observation. Will be updated in the next release. Thanks a lot.

TzOk83, 2024-06-01:
cps/mR/h to cpm/uSv/h conversion is 1 cps/mR/h = 6 cpm/uSv/h (you seem to use a value of 6.84 in your conversions), so 5 cps/mR/h = 30 cpm/uSv/h. After 24h background radiation comparisons with RKSB-104 (dual SBM-20 tubes, no averaging, just a simple rule, that counts during 180 sec. = nS/h) this is the best sensitivity value for my unit.

Gissio, 2024-06-01:
Ok!

closed 2024-07-01 by Gissio