Gissio/radpro #91: Error: init mode failed

JTknives, 2024-06-01:
I'm trying to flash an fs-5000. i have tried 2 separate counter with the same results. I can connect to it using the st-link utility and I included an image of that. but if i try to do anything with the utility it says device has shut down.

`Backing up old firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.566316
Error: init mode failed (unable to connect to the target)

Flashing new firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.564737
Error: init mode failed (unable to connect to the target)

Done.
Press any key to continue . . .`

[image]

ricsim, 2024-06-01:
You need to press the OK key all the time.
Gissio added this info to the FS5000 page.

closed 2024-06-01 by Gissio

JTknives, 2024-06-01:
I have been holding the ok/power button and it does not make a difference. I have tried it with the battery plugged in and unplugged. as soon as I hold the ok button the screen shows starting up and now the Geiger counter is powered on.

ricsim, 2024-06-01:
I had exactly the same issue. I never disconnected the battery.
Pushed OK and keep it pushed, then start the flashing while still holding OK pressed. (ignore that it turned on, just continue).
That worked for me.

JTknives, 2024-06-01:
When I do that I get the same thing. what's weird is it never shows 3v like all the other screen shots I have seen. mine always say 1.55v. And it says press any key to continue which I do and the window just closes. I bought the actual ST-link v2 not a clone.

Backing up old firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.551547
Error: init mode failed (unable to connect to the target)

Flashing new firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01557-gdd1758272-dirty (2024-04-02-07:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 500 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 1.551547
Error: init mode failed (unable to connect to the target)

Done.
Press any key to continue . . .

JTknives, 2024-06-01:
Here is a few pictures of my set up. I think everything is wired up correctly. 

[image]  

[image]

[image]

Gissio, 2024-06-04:
Try flashing with the Bosean FS-5000 powered through USB.

JTknives, 2024-06-11:
Yeah I tried all of that and still just say's 1.55v so I labeled it defective and returned to Amazon and ordered a $9 clone.