Gissio/radpro #93: GC-01 Has an entirely different board

mcalexer, 2024-06-04:
Hello all,

Never posted anything on Github before, so if I say something wrong let me know.

Just got a Fnirsi GC-01 in the mail a few days ago and was excited to get Radpro on it, unfortunately while reading the instructions, it seems like they have put out a total board revision. It is completely different, and even the piezo speaker is no longer on the back of the board.

To make things worse, the MCU is completely blank, they sanded off the identifying markings. Really blows, if I knew they did this, I never would have bought it.

Is there a certain MCU I should assume it has?

[image]
[image]

ReimuHakurei, 2024-06-04:
That's probably one of the counterfeit GC-01s and not a real one.

Gissio, 2024-06-04:
The board looks very similar to #48. Closing due to duplicate issue.

Gissio, 2024-06-04:
The board looks very similar to #48. I'd suggest returning it and get a refund.

Closing due to duplicate issue.

closed 2024-06-04 by Gissio