Gissio/radpro #103: Unknown hardware

Zendium0, 2024-06-17:
Hi there!
I ordered a FNIRSI GC-01 off Aliexpress to experiment with your fantastic firmware.
Upon teardown I ran into a completely different hardware to that described here on Githud.
No battery, no visible piezo and an unmarked IC and screen ribbon.

Am I completely out of luck?
Any idea as to which electronics it is and compatibility?

Thank you!!
Federico 

[image]

Gissio, 2024-06-17:
Please read #48.

closed 2024-06-17 by Gissio

Zendium0, 2024-06-17:
Thanks for the quick reply. 
I did have a suspension that my was the funky unit.
Difficult to spot the good units, FNIRSI doesn't seem to have an official store on Amazon.

Should you be interested I have found a data sheet for the chip version.

Thanks again!!
[CA32F0156系列MCU_产品介绍.pdf](https://github.com/user-attachments/files/15876983/CA32F0156.MCU_.pdf)