Gissio/radpro #106: GC01 issue

gc01noob, 2024-06-19:
Hi,
I have the APM version with J321 GM tube.
When it booted from new it said "ver 1.6" and read a background between 12-18uSv/hr

Windows 11 would not show the drive, USB 2 or 3 Pro or Home version no matter what buttons I tried. Tried on different PCs too, different hubs, ports etc. Device just boots with no drive created no matter what buttons pressed. eg Hold the power button and it boots then powers off.

In Ubuntu 24 it shows the BOOTLOADER drive when powered on - on any USB port.
Steps I took:
copied the RadPro 2 (later repeated with 2b3) - text file said READY.txt but no reboot, just showed 2 files on the drive.
red LED stays lit and wouldn't boot
removed the cable - also tried unmounting but made no difference
red LED stays lit and wouldn't boot
pressed reset button.
LED goes off but still wouldn't boot but did give BOOTLOADER drive with ERRAPP.txt file each time.

tried 100 different things eg every GC01 firmware, bootloader you shared to try and un-brick it.
sometimes the text in the drive says ERRAPP.txt sometimes it said READY.txt
always stayed constant red LED and required the reset button and a black screen.
LED was constant, longest I left it was 10mins.

copied the 1.6 version from the docs/device/GC01/firmware
2nd time i did this and pulled the cable, then pressed reset it finally booted

it now shows ver 1.6-2 when it boots
and my background readings vary from 16-58uSv/hr now.

I'm stumped!

any help would be appreciated as your firmware looks great but i'm scared I brick it again.

gc01noob, 2024-06-20:
observation 1:
the apm 1.6 bin is 63.5kb
radpro 2.0 is 147kb
could this be why it completely ignores the 2.0 bin file? is it too big?
no matter what i try it acts like there's no new bin file on the drive with 2.0

observation 2:
my device had a manufacturer sticker on the box showing it was a 2024 production batch, (the clock was set to 2022 and the LCD ribbon cable dated 2018) - could they have prevented the install of 3rd party firmware on the newer devices coming out the factory?

gc01noob, 2024-06-20:
note: the time/date and all settings were preserved all through this.
removing the 1220 battery only resets the clock back to 2022
removing the lithium does nothing
removing both does nothing but reset the clock back to 2022, all other settings remain intact.

I'm a bit worried now to remove both batteries and do a long-press on the reset button - this used to work on some other devices like motherboards to fully drain any flash but could also brick other devices for good.

As I don't have a programmer maybe I should just give up until someone else experiences this behaviour......or not.

thanks anyway.

gc01noob, 2024-06-20:
if I copy the 2.0rc5 file (112kb), then unmount and remount the text file changes to be "UNKNOWN.TXT"

2.0 often said "APPERR.TXT" when remounted

after reboot it will say "READY.TXT"

it always needs the reset pressed after any connection to turn off the LED and it then boots.

gc01noob, 2024-06-20:
observations:
if i copy radpro 2.0 over (READY.TXT) then unmount then remount (UNKNOWN.TXT) and copy again, unmount, remount (APPERR.TXT) copy again it will disconnect and brick the device. ie no boot just BOOTLOADER
to fix it i need to reset, mount (READY.TXT), copy 1.6, unmount, remount (UNKNOWN.TXT), then copy 1.6 again, it will then disconnect the drive, press reset to stop the LED and it boots into 1.6-2 again, keeping all settings and time.

I give up.

Gissio, 2024-06-21:
Try the [FNIRSI GC-01 alternative installation instructions](https://github.com/Gissio/radpro/blob/main/docs/devices/FNIRSI%20GC-01/install-stlink.md).

closed 2024-06-21 by Gissio

gc01noob, 2024-06-22:
not sure why you just closed this....anyway....

Here is how you install on Ubuntu 22.04 if you have the same issues I had above ie not seen in any Windows versions and copying the .bin does nothing, device shows black screen with constant LED....etc

1. rename the file to fnirsi-gc01-apm32f103rbt6-v2.0.bin (maybe not 100% required)
2. copy the file to the BOOTLOADER drive
3. leave the window open and copy it over again, confirm the overwrite

viola the GC01 restarts with RadPro 2.0.

it's literally that simple, so here's a wee trick that might be worth sharing with others "copy it twice" before going down the programmer/flashing route.

btw 2.0 seems to work perfectly, I'll try the 2.01 in a few days.

thanks.

gc01noob, 2024-06-22:
just to add - changed the tube to J321 from the default J613
my background is now back to what it was from the factory ie around 18-20 uSv/h (correct for my location)

great work btw it all really improves the device (will try PC logging next week)

thanks @Gissio

gc01noob, 2024-06-22:
for completion:
1. connect GC01 to Windows PC using quality USB-C cable, USB3.1 or 3.2 A port is fine
2. extract zip and open GiegerLog RadPro 1.43
2a. ignore the error message
3. click connect device
4. click Quick Log

live CPM data is displayed as it updates in the graph window

data displayed in NotePad:
==== Monitor Server Properties =======================================================
Autostarted at: 192.168.0.100 / 8080

==== Connect Rad Pro Device ==========================================================
Device successfully connected

==== Device Mappings =================================================================
Mappings as configured in GeigerLog's configuration file geigerlog.cfg.
Device    : CPM  CPS  CPM1st CPS1st CPM2nd CPS2nd CPM3rd CPS3rd Temp  Press Humid Xtra 
--------------------------------------------------------------------------------------
Rad Pro   : -    M    -      -      -      -      -      -      -     -     -     -    
Mapping is valid

==== Rad Pro Device Info Extended ====================================================
Configured Connection: Port:"COM4"
Device connected.             
Hardware ID                     : FNIRSI GC-01 (APM32F103CB)
Software ID                     : Rad Pro 2.0
Device ID                       : 796xx278
Device battery voltage          : 4.240V
Device time                     : 2024-06-22 22:20:25
Tube life time                  : 3:26:54
Tube life pulse count           : 5261
Tube rate                       : 31.750 CPM
Tube conversion factor          : 153.800 CPM/µSv/h
Tube dead time                  : 0.0876950 s
Tube dead-time compensation     : 0.0000000 s
Tube background compensation    : 0.000 CPM
Tube HV PWM frequency           : 47058.82 Hz
Tube HV PWM duty cycle          : 0.5000 %

==== Quick Log =======================================================================
Start logging using Quick Log database 'default.logdb'

==== Get Log =========================================================================
Log database: E:\temp\FNIRSI_GC01_geigercounter\geigerlog-radpro-1.4.3d-windows\data\default.logdb
LogFile newly created - available for writing

==== Start Logging ===================================================================
#DEVICES, 2024-06-22 22:21:49, Connected: Rad Pro : FNIRSI GC-01 (APM32F103CB);79615278 : CPS
#LOGGING, 2024-06-22 22:21:49, Start @Cycle: 1 sec

Windows 11 23H2 never sees the GC01 as a USB drive but all seems to work fine in GeigerLog

again thanks @Gissio  you've made this into a really useful device.

(if I don't update again it's safe to assume the 2.01 upgrade went well - I'll use the Ubuntu copy twice method from now on)