Gissio/radpro #107: GC-01 Overreporting

molisian, 2024-06-20:
Hi, I have installed the new 2.0 version on my GC-01 equipped with a J614 GM tube and compared to my stock Bosean FS-5000 with a much better J321 tube (better design and construction of the counter overall). When testing against a small source of Americium-241 or background levels the GC-01 tends to overreport the radiation level in Sv by almost double, sometimes even saturating completely and skyrocketing resulting in a "reset" (the level becomes -.--). Also, if it is possible, I suggest to make the adaptive fast algorithm faster, since compared to the FS-5000 it's much slower to detect a source. Or maybe create an option called adaptive super fast idk. Just a thought, I really appreciate the developer work :)

Gissio, 2024-06-21:
Can you make a video showcasing the response time of the Bosean FS-5000 and Rad Pro?

molisian, 2024-06-21:
Here are the videos, remember I have not installed radpro on the FS-5000, only on the GC-01. https://1drv.ms/f/s!AtXtrPDofI3fnF6hsrigzKZNjMyS
The conversion factor I'm using is the standard one for the J614 tube in the menu.

Gissio, 2024-06-21:
Regarding the slower response of the GC-01, this behavior is expected. The two devices you are comparing have tubes with significantly different conversion factors. Under identical conditions, the device with the lower conversion factor will receive fewer pulses per second than the other. As a result, it will take longer to average.

Regarding the selected conversion factor of the J613 tube, I insist that you continue this conversation in #89.

closed 2024-06-22 by Gissio