Gissio/radpro #111: RadPro History scale on GC01

gc01noob, 2024-06-24:
GC01 running RadPro 2.01 - Dusk Theme
the 3 History graphs are:
1. 10mins - scale 0.01 - 1 uSV/h - no middle line
2. 1h - scale 0.01 - 1 uSv/h - middle line
3. 24h - scale 0.1 - 1 uSv/h - no middle line

there is a middle line on the 1h graph but unfortunatley this does not correspond to the middle of the scale (0.5) so there's no way of reading the graph and determining, even roughly, what the readings were. looks like ~0.16 is on the line and ~0.3 is half way up the top part.

is there any way to improve the scale? even making the middle line equal to half way between the scale would be great. (or any approx. indication of the readings on the graph)
split into 10 lines would be amazing eg 0.1 graduations or 5 lines with 0.2 uSV/h per line.

closed 2024-07-01 by Gissio

gc01noob, 2024-07-01:
even the original firmware manages to put a scale on the history graph.....
now I understand why people still solder on the capacitor to get a click rather than use this firmware.
oh well, hopefully someone will fork it and fix it....

FacuM, 2024-07-01:
> even the original firmware manages to put a scale on the history graph..... now I understand why people still solder on the capacitor to get a click rather than use this firmware. oh well, hopefully someone will fork it and fix it....

Lol, did you even donate money to the owner or something to begin with?

gc01noob, 2024-07-01:
1. it's FOSS
2. if he doesn't care and doesn't mind putting out low quality code without basic functionality then good for him. All this firmware does in reality is click.
2a. there's not a chance in hell I'd pay him now.
3. if you use my discovered and shared working method for flashing the firmware when Windows 10/11 fails that actually works and doesn't risk bricking it or require the purchase of a programmer ie using Linux and copying twice using the GUI then you owe me $10... 

lmfao gfy.

FacuM, 2024-07-01:
> 1. it's FOSS
> 2. if he doesn't care and doesn't mind putting out low quality code without basic functionality then good for him. All this firmware does in reality is click.
>    2a. there's not a [...]

Just FYI, the latest official firmware **does** support making clicking sound. No need for hardware modding. You can find it in this repo.

gc01noob, 2024-07-01:
where is it? what version? 1.6-2 or later?

FacuM, 2024-07-01:
> where is it? what version? 1.6-2 or later?

Yeah, 1.6.X IIRC.

gc01noob, 2024-07-01:
ok didn't notice that when i installed it, guess I'll try again to see
the Linux copy twice trick makes it so easy to swap firmwares it would be simple to use both, 1 for geigerlog use with a laptop/Pi and the factory for outside in the wild. (shame though as a basic scale on the histpry graph would be such a great UX benefit to RadPro)
cheers.

FacuM, 2024-07-01:
> ok didn't notice that when i installed it, guess I'll try again to see the Linux copy twice trick makes it so easy to swap firmwares it would be simple to use both, 1 for geigerlog use with a [...]

Sorry, I forgot to tell you the path.

docs/devices/FNIRSI GC-01
/firmware

Here you go!

Gissio, 2024-07-07:
The plot is logarithmic. This means every division represents a tenfold increase in radiation.

gc01noob, 2024-07-07:
can confrim that with firmware 2.0.2 my device now displays history perfectly as a log scale ie each horizontal line is a factor of 10.

thanks Gissio