Gissio/radpro #128: Unable to set the date/time

morfikov, 2024-07-19:
I just found this project and gave it a try, because my GC-01 just stopped working for unknown reason (when turned on it just make a noisy beeep sound, there's no way to turn it off). 

I flashed the radpro firmware and my GC-01 backed to life.

The problem I see with the radpro firmware is that I wasn't able to set the time/date. When I try to do so, the device reboots and when it's booted again, the time/date is not set.

MCU: CH32F103R8T6

dc1rdb, 2024-07-19:
You might want to check the button battery for the real time clock. Looks like it is depleted...

morfikov, 2024-07-19:
Yes it was, but yesterday I replaced it with a brand new one.

Gissio, 2024-07-25:
Since you report your device is defective, I doubt the issue is related to Rad Pro.

closed 2024-07-25 by Gissio