Gissio/radpro #130: Werid radiation readings

morfikov, 2024-07-20:
I've been testing the firmware for a day, and something is wrong. When the device is booted, everything seems to be good. But after a minute or two, the readings raise from 0,2uSv/h to 100-200uSv/h. And no, I'm not in the vicinity of Chernobyl. :] When I covered the left side of the GC-01 counter with my hand, the readings dropped to normal. When I uncovered it, the device seemed to work fine for a while, and after a minute or two it started to show 100-200uSv/h again. Does anyone have a similar issue?

jasperweiss, 2024-07-20:
A picture of the inside might make it easier to tell what's going on. What tube does it have? If the seller replaced it with a cheaper one, which is what most sellers do, unless you bought it from F-NIRSI directly, they might not have bothered to change the voltage level to something appropriate for that tube. You need to set the proper conversion factor as well (but it's not going to solve the spiking).

gc01noob, 2024-07-20:
the first GC01 I purchased did similar, it showed readings that would mean that my entire town would be cooked within minutes (over 300 uSv/h). i returned it.
second one works perfectly.
If it did work before the firmware upgrade check that:
1. you have selected the correct tube in the settings
2. you use the correct firmware for your CPU - try a reflash back to 1.6 and back again
3. if above is all ok, make/buy a high voltage probe and check the tube voltage - if it's too high you can either solder on a reducing resistor or create a new profile to reduce the voltage - there's a recent discussion on how to do this.

morfikov, 2024-07-20:
It's directly from F-NIRSI and with the stock firmware it was working well for over a year. With radpro it was also working well for a few hours, but suddenly it started to acct weird.  

Here are the photo: https://imgur.com/a/IthGvlL

dc1rdb, 2024-07-20:
Stock firmware at https://github.com/Gissio/radpro/tree/main/docs/devices/FNIRSI%20GC-01/firmware

Your tube voltage is way too low. Should be between 360 and 440 Volts.

morfikov, 2024-07-20:
The stock firmware acts the same. So apparently something is wrong with the device.

Gissio, 2024-07-25:
Since you report the original firmware presents the same issue, I doubt the issue is related to Rad Pro.

closed 2024-07-25 by Gissio