Gissio/radpro #142: version 2.0.2: FS2011 does not start up

wim04, 2024-09-22:
version 2.0.2: FS2011 does not start up.
CPU: GD32 F 150 C8.
Flached with STLINK. Back to v2.0 and the FS2011 starts up.

Gissio, 2024-09-22:
Thanks for the report! I'll take a look at this once I find the time.

wim04, 2024-09-23:
The good news: version 201 and 202 starts up. There was a break in the power cable.
But neither of these versions shows any data. Menu and backlight is working.
Only version 2.0 works for me.

Gissio, 2024-09-25:
Could you please make a video showcasing the error and upload it here? It would be tremendously useful.

Dsl71, 2024-10-08:
Hi, I have 2.0.2 running fine on mine (same CPU, SBM20-U mod), FYI.

Gissio, 2024-10-08:
Thanks for the report.

Closing this issue, as its nature seems more specific than general.

closed 2024-10-08 by Gissio

wim04, 2024-10-20:
Hi Gissio,

Sorry for the late reply!

There were 3 problems after flashing to version 2.02

- Brightness set to minimum
- Contrast set to minimum, When Contrast is set to minimum, nothing is readable on the screen
- HV profile was Custom instead of Default, result, no data on screen

Solution Contrast:
Turn on device
Push menu/ok
7x Down ( = Display)
menu/ok
1x Down  (= Contrast)
menu/ok
4x Down (Contrast level 5)
menu/ok
3x  >

2- No data on Display
Cause:
HV profile was Custom, result: no data on screen
Solution Change HV profile to Factory default
Tube J305

After that, version is ok and i can read the screen.

Greetings,
Wim

Gissio, 2024-10-20:
Most likely this happened because you flashed the 2.0.2 upgrade on top of the 2.0 release.

Let me point to the 2.0.2 installation instructions: https://github.com/Gissio/radpro/releases/tag/2.0.2