Gissio/radpro #143: GC-01 unable to connect

Urminme, 2024-10-01:
I have tired connecting The GC-01 to my computer and when I plug it in I get a windows message stating there is an issue with the USB device, tried with several cables, I opened Device manager and when I hold power down it does seem to have something change on Device manager as it resets but im just not sure what it is. Do I need to have some specific driver or is there something wrong with my Geiger counter?

ChyrkaA, 2024-10-01:
No, everything is fine with your device. It needs to be reflashed, most likely through a programmer. Or from Ubuntu. I had the same situation, you can read in this thread:
https://github.com/Gissio/radpro/discussions/133

Urminme, 2024-10-04:
> No, everything is fine with your device. It needs to be reflashed, most likely through a programmer. Or from Ubuntu. I had the same situation, you can read in this thread: #133

So I was able to get it programmed and had to use the ST-link method unfortunately, but im not sure if I didnt set it right and my counter is all over the place it never dips below 0.500 uSv/h and often goes way up to over 1.000uSv/h which is not possible there is nothing that would give off that much everywhere I go, is there a way to tell if my tube is bad or something

dc1rdb, 2024-10-04:
Your HV might be too high. I would suggest you go to a custom HV profile in your Geiger settings and start with a real low value, like 5 kHz and 1% PWM duty cycle. Then check if your counter readings stabilize at normal values.

Urminme, 2024-10-04:
> Your HV might be too high. I would suggest you go to a custom HV profile in your Geiger settings and start with a real low value, like 5 kHz and 1% PWM duty cycle. Then check if your counter readings stabilize at normal values.

Thank you for the help, so I tried 5kHz and 1% but it just did nothing after that no measurements, I went back to the standard and it started working again, its weird the meter is always measuring high and sporadic no matter where I am at or what im doing, sometimes its lower at 0.300 and randomly jumps up to 1.300 then back down to 0.600 its just all over the place, Ive tried messing with background compensation which does not make much difference. It just seems horribly inaccurate and im very doubtful its Radpro as it was very squirrely before I switched over to this much much better firmware, I was hoping this would make things more accurate and stable but im not sure what else to assume other then the tube is just shitty or something else is causing erroneous readings, its using a M4101 tube and set to such in the geiger settings

dc1rdb, 2024-10-05:
No offense, but have you double checked correct polarity of your geiger tube? (I had that problem once)

Gissio, 2024-10-06:
Converting to discussion, as specified in #32.