Gissio/radpro #144: Different tube

J30789, 2024-10-04:
Hello I have the FNIRSI GC-01. Opening it up I have a different tube to the ones you listed. On the tube (pictured) it’s a GM4011H24.05 on the board you can see it says J613 but thought I check before I do anything else. The MCU is APM32F103RBT6 and the screen ribbon is ENH-TV0240b026

thanks in advance 
[image]
[image]
[image]

jonwhite227, 2024-10-04:
It's an M4011 made in May 2024. It's equivalent to the J321, so the correct tube for the GC-01, and far better than the J613. Nothing to worry about.

J30789, 2024-10-04:
Love it! Thank you!

closed 2024-10-04 by J30789