Gissio/radpro #151: Black screen on startup Radpro does not start (Intermittent Problem)

aprusek, 2024-11-07:
I have two FNRSI GC01 detectors.
A Blue one with a J321 (which needs replacing) and a Yellow one with a M4011.
Radpro Version 2.0.2

Radpro installed on the Blue unit works fine.
Multiple power cycles over multiple days no issue.

Radpro installed on the Yellow unit requires re-flash after a few days.
When working all seems well.
When problem appears Radpro fails to display a screen and no clicking sounds on powerup.
Once this occurs, power cycles do not change this behaviour.
Unit will power down and go into bootloader mode.
Loading software via bootloader, either FNIRSI or Radpro brings unit back to life.

FacuM, 2024-11-07:
Honestly, these symptoms sound just like what a bad flash storage would do (very high chance of charge decay). I'm gonna guess your only chance is to buy another one or get a replacement board if you can manage to find one.

Gissio, 2024-11-21:
Closing issue, as it appears to be a device-related problem.

closed 2024-11-21 by Gissio