Gissio/radpro #163: Minor graphical glitches at startup

ThatBum, 2024-12-06:
I first noticed this in the latest build of 2.0.4. On the splash screen there's a line of corrupted pixels here:
[image]

They're not dead pixels on the display, once it's past the splash screen it goes away.

On 2.0.3 the splash screen disappears too quickly to see if it's there, but I did notice a similar bit of corruption on the main screen:
[image]

This also goes away after a second. I guess something with the video memory isn't initializing quite right.

Unit is a FNIRSI GC-01 with the Geehy processor and an M4011 tube. If it helps any, the code on the display ribbon cable is ENH-TV240B026.

ugopapetti, 2024-12-06:
Also noticed the same dots.

FNIRSI GC-01, APM32F103RBT6 - 2418, M4011, display flat cable: GMT024-05
V2.0

Is this an issue? However...

ugo

ThatBum, 2024-12-06:
> Is this an issue?
Thus the minor in the title. I know it's probably low priority but figured I should report it anyway. Might be a trivial fix.

ugopapetti, 2024-12-06:
agree

Gissio, 2024-12-06:
Thanks for the report!

¿Can somebody confirm this did not occur in 2.0.2?

ThatBum, 2024-12-07:
> Thanks for the report!
> 
> ¿Can somebody confirm this did not occur in 2.0.2?

It's not on the splash screen like 2.0.4, but does appear on the measurement screen for a second like 2.0.3.

Gissio, 2024-12-07:
In 2.0.4, does the spot always appear at the same spot, or does it move every time you turn on the device? Does the spot's pattern stay the same or does it change (another photo might help)? Is there any spot afterwards?

I insist: could you check if there is any spot in 2.0.2? This would help a lot in diagnosing the problem.

ThatBum, 2024-12-07:
> It's not on the splash screen like 2.0.4, but does appear on the measurement screen for a second like 2.0.3.

This is in 2.0.2, that's what I meant by this.

The glitch has the same appearance and the same position every time.

Gissio, 2024-12-14:
Can somebody check if this also manifests on the Bosean FS5000? Trying the landscape release would help a lot.

tahinchik, 2024-12-16:
On 2.0.3 i have the same issue only in startup screen. And its happened so fast (milliseconds) but  its not affects
[image]

Neptuns, 2024-12-19:
This is on 2.0.2. Dotted line always appear on startup. Double dots above decimal separator always appear on first measurement update. The small dots after number 4 appear rarely and randomly.
[image]

ugopapetti, 2024-12-20:
Here my dots with 2.0.4:

ugopapetti, 2024-12-20:
Hi all,
so,me tests with the counter GC-01
Processor brand / moidel: Geehy APM32F103RBT6 - 2418
Geiger tube: GM4011H
Display model: GMT024-05 V2.0
Internal port layout: #VCC – GND – DIO – CLOCK (VCC: square)

2.0.2: splash screen without corruptions, measurement screen corrupted for
about 1,5 seconds, then screen becomes as expected, without any extra dots.
2.0.3: splash screen too fast to observe for defects, measurement screen
corrupted for about 1,5 seconds, then screen becomes as expected, without
any extra dots.
2.0.4: splash screen corrupted,  measurement screen ok

Neodymium5312, 2025-01-08:
I can confirm that I am also having the same issue, I also am sure that I have official gc-01 device, came with j321 and it has flashed to newer official firmware successfully in the past. I'm on 2.0.3, everything else seems to function correctly otherwise

Processor brand / model: Geehy APM32F103RBT6 NT4CP 5090 D1 - 2234
Geiger tube: J321
Display model: ENH-TV0240
[image]
1B026 V0.3
Internal port layout: #VCC – GND – DIO – CLOCK (VCC: square)

Gissio, 2025-01-14:
Can somebody try this: https://github.com/Gissio/radpro/actions/runs/12775724387 (click below on radpro-firmware-snapshot)?

ThatBum, 2025-01-14:
> Can somebody try this: https://github.com/Gissio/radpro/actions/runs/12775724387 (click below on radpro-firmware-snapshot) and see if it fixes this issue?

White screen. It is detecting counts but is otherwise unresponsive.
[image]

Neodymium5312, 2025-01-14:
> > Can somebody try this: https://github.com/Gissio/radpro/actions/runs/12775724387 (click below on radpro-firmware-snapshot) and see if it fixes this issue?
> 
> White screen. It is detecting counts but is otherwise unresponsive. 

Same issue here as well with the snapshot, nothing but blank display with backlight but can hear clicks and see led activity

Gissio, 2025-01-15:
Thanks for the reports.

There was a very silly bug. Could you please try again? https://github.com/Gissio/radpro/actions/runs/12780156091

ThatBum, 2025-01-15:
> Thanks for the reports.
> 
> There was a very silly bug. Could you please try again? https://github.com/Gissio/radpro/actions/runs/12780156091

This one works and the artifact is gone from the splash screen but this one is still present.
[image]

ThatBum, 2025-01-15:
Probably unrelated but in this new build I am also getting a weird font artifact. It's present when there's "14" anywhere in the reading but nowhere else as far as I can tell.
[image]

Gissio, 2025-01-16:
Those images were very useful!

Can you try once again? https://github.com/Gissio/radpro/actions/runs/12816891832

Neodymium5312, 2025-01-16:
> Those images were very useful! I was able to reproduce and fix the glitch involving the 1 and 4 characters.
> 
> Can you try once again? https://github.com/Gissio/radpro/actions/runs/12816891832

This one seems to function properly 
[image]

ThatBum, 2025-01-17:
The Pulses menu is messed up. Haptic leads to Pulse LED, Pulse LED leads to Display flashes, and Display flashes leads to Haptic.

Gissio, 2025-02-03:
Can you try https://github.com/Gissio/radpro/actions/runs/13103763916/artifacts/2524185828 and see if this is fixed?

Gissio, 2025-02-03:
Excellent bug report. Thanks for your help.

Gissio, 2025-02-03:
Excellent bug report. Thanks for your help.

Did you try the new features? Alarm dismissal, more rate/dose alarm options, configurable alarm signaling, keyboard lock mode?

Do the graphical glitches still appear?

ThatBum, 2025-02-03:
> Excellent bug report. Thanks for your help.

Thanks, I'm told I have a knack for finding bugs. Speaking of, another more serious one. It seems the cumulative dose is not saved upon power cycling anymore. Every time it starts with zero counts.

The new features all work, however I'd like to suggest a change with alarm signaling. The way it is now the Acoustic setting applies all three levels of alarm, Haptic applies itself and Visual, and Visual just applies itself. You can only choose one of the three as if they were radio buttons. Instead I feel it would make more sense to be able to select any combination of these such that one could apply them individually. Furthermore I'd like an LED alarm option because I'd prefer not to have the alarm flashing obscure blinks from the pulse counts.

The glitch is still there. I had no idea this would be such a bugbear to hunt down!

ugopapetti, 2025-02-03:
Hi Gissio,

Glitches @splash are not present fir me. Instrad for 1 secondo this apparso
After Power on, as in the attached picture.

Thanks for the developments, keyboard lock prevents to enter config mode,
alarms, Need time to test them.

Cheers ugo

Gissio, 2025-02-03:
Ugo: I don't understand what you mean:

> Glitches [@Splash](https://github.com/Splash) are not present fir me. Instrad for 1 secondo this apparso After Power on, as in the attached picture."

ugopapetti, 2025-02-03:
Gissio, saw some mispelled words due to the Word corrector of my mobile...

After Power on the screen with the release Is without any glitch.
When the measurement screen apparso there Is a corruption, as in the
picture below, above the decimali point.
The dots remains for 1 secondo then disappear.

Cheers ugo

Gissio, 2025-02-09:
There was an error with the watchdog timer (which resets the counter if the system freezes). Could you try again with 2.1beta5? https://github.com/Gissio/radpro/actions/runs/13228168353/artifacts/2561063487

ThatBum, 2025-02-11:
The transient glitch on the main screen mentioned by ugo is still there ([this one](https://github.com/user-attachments/assets/02f71662-a3b6-44c6-819b-3c319a6e2887)), but everything else above seems to be fixed. I appreciate my feedback being implemented so quickly!

Couple more nitpicks for you:

- When the device is plugged in to power and turned off, and the battery reaches full state of charge, the green LED doesn't turn on. The red LED just dims. The green LED does turn on when the device is at full SoC, plugged in, and turned on. This happens with the stock firmware as well. This may not be fixable because the charge LEDs might be controlled by a battery management IC or other low-level features, but I figured I'd mention it.
- On the statistics page, why not put the lowest observed dead time value into persistent memory? That way the measurement will get increasingly accurate with more runtime without it resetting every session.

This is looking like it'll be the most polished release yet!

Gissio, 2025-02-11:
ThatBum: thanks for the feedback.

Here's how the battery indicator bars work:

0%: 0-20% charge
25%: 20-40% charge
50%: 40-60% charge
75%: 60-80% charge
100%: 80-100% charge

So, when the battery indicator goes to 100% while charging, it actually means the battery is at 80% capacity.

The red and green LEDs are controlled by hardware. The LED turns green when the battery is fully charged.

So, I guess we are ready for the Rad Pro 2.1. Should anyone present know of any other issues, speak now or forever hold your peace.

ThatBum, 2025-02-11:
> The red and green LEDs are controlled by hardware. The LED turns green when the battery is fully charged.

Yeah that's what I figured. Still weird that it only works correctly while it's on.

But what about storing the dead time value, is there a reason I'm not thinking of that this is a bad idea?

Gissio, 2025-03-11:
Can you try https://github.com/Gissio/radpro/actions/runs/13797030926/artifacts/2732897104 and see if the issue is fixed?

Gissio, 2025-03-26:
I guess this still happens with the latest beta:

https://github.com/Gissio/radpro/actions/runs/14044292193

Gissio, 2025-03-29:
Can you confirm this still happens with the latest beta?

https://github.com/Gissio/radpro/actions/runs/14044292193

gfMX, 2025-06-14:
Graphical glitch in 2.1.1 (CH32F103)

[image]

[image]

Gissio, 2025-06-15:
That’s quite bad. Does this also occur when you connect a powered-off GC-01 to a USB port?

> [@Gissio](https://github.com/Gissio) it happens every time I turn it On. But it seems only a graphical glitch, after that everything works fine

gfMX, 2025-06-15:
No it does not happen when connected to the USB. Also reflashed today the firmware and the Glitch it's gone. 
The only thing left are the lines or dots when it is starting (second image)

Gissio, 2025-06-23:
Can somebody else confirm there is no graphical glitch when the device is turned on through USB?

dc1rdb, 2025-06-23:
Geehy GC-01 with 2.2 beta 2 here. No graphical glitch when a powered-off GC-01 is connected to a USB port.

Gissio, 2025-06-23:
> Geehy GC-01 with 2.2 beta 2 here. No graphical glitch when a powered-off GC-01 is connected to a USB port.

That’s great news! Thanks for the report.

Gissio, 2025-08-30:
Does this glitch still occur in 3.0?

gfMX, 2025-08-30:
Yeap, still happening in 3.0

[image]

Gissio, 2026-01-30:
TL;DR: The problem was caused by the device’s bootloader not initializing the stack pointer correctly. This led to global variable corruption during the C runtime startup sequence. Fixed by taking full control of the reset handler.

The correction will be included in the upcoming Rad Pro 3.1 beta release.

Thanks to everyone who helped report the issue, much appreciated!

Gissio, 2026-02-07:
Could you check if 3.1beta1 fixes this?

jdogg836, 2026-02-08:
> Could you check if 3.1beta1 fixes this?

Confirmed no graphical glitch on a GC-01 w/ ch32f103r8.

closed 2026-02-08 by Gissio