Gissio/radpro #173: Surprise GC-01

ullix, 2024-12-19:
Looks like "FNIRSI GC-01" is the thing to sell, irrespective of what you put inside. AliExpress sells for as low as 14.75€ a device clearly labelled as  "FNIRSI GC-01". 

[image]

For that price, I thought, it is good to have a spare, so I ordered one and got a BIG surprise. The front side looks very familiar; the GUI too. GUI works as before.

[image]

The value of CPS=4 is strange. With my setup I should have seen something between CPS=20...60! 

Looking at the inside I saw a circuit board completely different from the GC-01 I have seen so far:

[image]

The tube has no markings, but looks like a j613. It is smaller than the M4011, J321, J305 and similar, but that size difference is too small to explain the rather huge difference observed in sensititivity.

The MCU also comes with markings removed. The HV generator is probably where the "103" (aka 10mH) inductivity is, but also looks oddly different from other GC-01.

The battery (no markings) is a round-cell. I had charged it overnight, but the device would not even start with it - shutting down in a 10 sec countdown - requiring a USB connection for startup. Once it was running I could remove the USB cable, and it kept running for a while (after 1h the batt indicator showed half-full in green color).

Of course, this thing needs RadPro on it! Next surprise: Neither on Win11 nor on Linux debian12 could I see a connection, neither when connecting after the "GC-01" is switched on, nor when connecting first and then switching on. Dmesg did not see any USB connection!

I suspect the USB data lines are not even connected; it serves only for power supply? Not sure how to test this.

Without RadPro this is just electric trash. Any suggestion on how to rescue this Pseudo-"GC-01"?

ThatBum, 2024-12-20:
This is a counterfeit. See #48, #93

ullix, 2024-12-20:
@ThatBum Thanks! Good Lord, are they selling that many GC-01 devices that it is worth to make such a low cost fake?

The MCU is: CA32F0156L2-68  (is displayed when device is USB-connected and then the up-arrow is long pressed). However, not of much help to me.

ThatBum, 2024-12-20:
FNIRSI is actually a reasonably prominent name, believe it or not, especially in China. They make all sorts of test gear. So it'd be reasonable for someone to try and snipe some of their customer base. See https://www.fnirsi.com/

I don't have any evidence of this, but Rad Pro might have brought about these counterfeits' existence. The GC-01 has been especially popular as of late because it's the only supported model that can flash the firmware over USB and not have to faff about with ST-LINK.

ullix, 2024-12-20:
Just to confirm previous observation: I got these test connectors ( https://www.amazon.de/dp/B0BXKS3B8M) and checked the (many) USB-C pins. No, no activity on any of them, except on the Power pins. This plug is solely for power supply; it has no other function.

soosp, 2024-12-23:
The 6 pin USB-C sockets are for power delivery only. It has not D+ and D- pins for data transmission.

ullix, 2024-12-25:
@soosp while true for this specific device, it surely is **not true for the original device**, of which there are at least two variants. And who knows what the other fake GC-01s will exhibit, that we will be seeing in the future!

soosp, 2024-12-25:
@ullix You are right. I talked about the USB-C socket seen on the picture above. It have six pins, and it designed for power delivery only. My point was that this counterfeit one will definitely not communicate over USB. Sorry if it was not clear.

Gissio, 2024-12-29:
Closing as this is a duplicate of https://github.com/Gissio/radpro/issues/48, https://github.com/Gissio/radpro/issues/93 and https://github.com/Gissio/radpro/issues/103.

closed 2024-12-29 by Gissio