Gissio/radpro #177: FS-5000 Radpro 2.03 wrong Landscape orientation

Memorandlum, 2024-12-28:
1.) 
  You´ll never want be between a radioactive source and your measuring device and look over the source to read the display or serve
  the device by moving around the source. The display's orientation has to be turned 180 degrees, then, the fundamental Rule 
  "ever to be as far as possible away from any kind of radioactive source" is retained.

First of all I appreciate radpro a lot, thx for your ambition and work!
Second, well.... there's the need to do some inevitable but easy to do changes and additions to radpro to become a more
serious(!) professional(!) and safe(!!!) tool as well. The design of cheap measuring devices maybe leads itselfe to some restrictions in the kind of use but the best device is the one you have.

Gissio, 2024-12-29:
I don't follow you.

Memorandlum, 2025-01-08:
I'm sorry I'll try again. :-)

I had to write the orientation onto a sheet of paper to demonstrate the problem because at the moment 
I don't use the landscape firmware on my FS-5000. 
Perhaps a 180° rotateable landcape-display would make sense while playing around with active stuff.
With the actual landscape firmware, the best way to use it is to flip the device upright and above the ground.

[image]
[image]

Gissio, 2025-01-11:
Rad Pro's landscape orientation ensures that the up and down buttons remain consistent.

If you perform field measurements, I suggest you get a telescopic pole with a phone holder to safely distance the counter from you.

If you perform experiments at home, I suggest you manipulate the counter with an appropriate pole.

closed 2025-01-11 by Gissio