Gissio/radpro #178: FS-5000 Radpro 2.03 program variables initiate problem

Memorandlum, 2024-12-28:
If you turn off the device at Radpro Screen "Average" and reactivate the device very quick, the old measurements are still
present. If you wait several seconds while off, then the data is gone. That´s an inconsitantcy and could lead to an unpredictable
behavier of the device. All used temporal program variables and Data has to be set (initiated) to a defined value status 
at every device-startup!

Gissio, 2024-12-29:
Thanks for the report.

One request: on your FS5000 do you get to see display glitches as reported here: https://github.com/Gissio/radpro/issues/163 ?

Memorandlum, 2025-01-12:
FS-5000 radpro2.03
nope, everthing is fine, no glitches, no corrupted pixels. 

I felt that the display refresh in the landscape revision was somehow different from the upright revision, 
I barely tested it but I think it was slower. I will do more detailed testings, when you'll puplish the revision 2.04 
like the speed of response for an alarm.

Gissio, 2025-02-03:
Can you try https://github.com/Gissio/radpro/actions/runs/13103763916/artifacts/2524185828 and see if this is fixed?

closed 2025-02-09 by Gissio