Gissio/radpro #179: FS-5000 Radpro 2.03 dosimetry problem

Memorandlum, 2024-12-28:
I like the easy and fast use of Radpro but at the moment, that leads to a serious disadvantage, You musn't(!!!) use it for dosimetry if you
enter a highly radioactive zone like Tschernobyl. (at the moment..... unthinkable!)

The device's physical design doesn't prevent you from switching of or resetting collected data accidentically. Then your health and life is in real danger, seriously. Dataloss by fault or accident should never be possible during dosimetry situations! 
(btw No Backup....No Mercy) 

There should be a "lock" in the config-menu or a "Dosimetry" point, where you can lock(prevent) Data from beeing deleted for: 
"History", "Instaneous", "Average", "Cumulative" (default on) and the "Power-off-Switch"
The Power-Switch should get a doubled safed lock because when the device is off, you dont´t get any alarms. By the way,
Sound is broken (klicks/beeps) after an alarm had occured.
And when the longpress-delete-funktion ist obsolet, you have a new key for four new functions at the four present screens. ;-)

gc01noob, 2024-12-28:
just a passing 2cent comment but if your life depends on it at military or nuclear disaster zones maybe buy a professional device..... no one sensible is using a cheap, uncalibarated device to protect their life at that level of risk. it's a basic device for rough readings or simple indication of some present risk, i'd even argue that some occasional hospital use would be ok.
imo spend the money if it matters.

Memorandlum, 2024-12-28:
Well, if nothing matters there's no reason to get too excited, too. 
I think there are only little changes needed to be done to get a better result and a more useful device.
Maybe a hint in the Radon Statistics-menu "Radon rev. X.xx" and this github-side because the startup screen 
disappears after about 2cent milliseconds. Well criticism should always to be gladly seen, to criticize isn't. ;-)

NuclearPhoenixx, 2024-12-28:
A simple "lock" state for the device that can be turned on or off by a specific key combination would probably solve this "issue", right?
I'm agreeing with @Memorandlum, because this is not a professional device in **any** shape or form. I don't think this is a large safety risk or an issue, but more of a "nice to have".

Gissio, 2024-12-29:
Thanks for the suggestion.

One request: on your FS5000 do you get to see display glitches as reportes here: #163?