Gissio/radpro #195: Display Colour - Enhancement

sivaelid, 2025-01-11:
Very nice software thanks for the development..
The Original software made more use of colour screen  the white/blue  is very minimalist and is not so good in strong light, Consideration as an enhancement  to change the colour  to say yellow text on Black background or other high contrast display setting. This may also save some power and give  a better battery life.

S Collins

Neodymium5312, 2025-01-11:
I'd also greatly appreciate an option for colorization, maybe something along the lines of a few options such as "monochrome" or "colorized graphs" as alternatives, this was suggested by another user who was doing reviews of the gc-01 for graphing and looks pretty neat as a feature, also colorizing on the "instantanoues" screen could be done as a percentage of the "max" setting such as yellow for 50% of max set cpm and anything above 70% cpm could be red just as an example idea.
[image]

Gissio, 2025-01-11:
Rad Pro follows the [OpenBridge](https://www.openbridge.no/) UI design system.

closed 2025-01-11 by Gissio