Gissio/radpro #197: Unable to connect Geigerlog-RadPro to RadPro for data download with V2.03

sivaelid, 2025-01-27:
HI Great software on the latest version 2.03. I have tried to use the Geigerlog-RadPro to access the stats but get  the  response below.

 I am using the correct  setting you describe for the com-port and Windows has  found correctly the FNIRSI GC-01/JOY-IT JT-RAD01 with a APM32F103R8 (Geehy) processor  and as a valid comport  on the windows machine.  Is this an Issue due my lack of knowledge, or is it a Bug ? or is Geigerlog- RadPro not comparable  with version 2.03?

27 19:56:58.441 DEBUG  : .....1    EXCEPTION: CANNOT initialize Monitor Server at Port:8080 ([WinError 10013] An attempt was made to access a socket in a way forbidden by its access permissions) in file: gweb_monserv.py in line: 118

Thanks again for this great  software development, and all the work you have put in on the design.

Simon

gc01noob, 2025-01-27:
have you checked that port 8080 is open in the firewall? IN and OUT?
Windows firewall can be a real PITA sometimes - eg if you are on the Public setting it can block many extra ports.

gc01noob, 2025-01-27:
old win7 but might help
https://stackoverflow.com/questions/2778840/socket-error-errno-10013-an-attempt-was-made-to-access-a-socket-in-a-way-forb

python related:
https://github.com/googleworkspace/python-samples/issues/86

https://community.mech-mind.com/t/topic/1913

try a serach for "WinError 10013" all point to either firewall or already used ports.

gc01noob, 2025-01-27:
netstat and other basci networkign tools might help as it might even be as simple as you have it running twice or another program is using the port

sivaelid, 2025-01-28:
I have tried the suggestions above,  still get the same error. I  can connect to the unit via a serial terminal and get correct response for all commands  like data retrieval.

Trying to find gweb_monserv.py  in version 1.4.3 c/d it does not exist. I have looked in version 1.5 on the geigerlog site code and this file exists there. 

I see also on the Geigerlog-RADPRO  issues  list, someone has put that they can not connect to GC-01, so may be an issues. In my case I fined the external IP connections and correct port numbers  and all port are allowed though firewall etc, but the issue is the connection from Geigerlog to  the device comport, it just hangs  and then have to cancel the geigerlog program.

I hope this helps , I wonder does any one with a GC-01 got the Geigerlog program working ?

gc01noob, 2025-01-28:
I use a GC01 but running 2.02 still, I don't use windows firewall and it works perfectly.
sorry can't be more of a help but that error is a common windows firewall error code. (did you try to fully disable it?)

sivaelid, 2025-01-28:
Hi thanks for your reply and you using 2.02 on GC-01. I will try that,  as yes check firewall open and also with it disabled, I check other people  what they did with windows error and also cleaned the DNS  entries and  tested the rout external access. I will carry on looking for a solution.

I am very pleased with the GC-01 with Gissio - RadPro software.

Thanks for your help.

Simon
________________________________
From: gc01noob
Sent: Tuesday, January 28, 2025 16:19
To: Gissio/radpro
Cc: Si; Author
Subject: Re: [Gissio/radpro] Unable to connect Geigerlog-RadPro to RadPro for data download with V2.03 (Issue #197)

I use a GC01 but running 2.02 still, I don't use windows firewall and it works perfectly.
sorry can't be more of a help but that error is a common windows firewall error code. (did you try to fully disable it?)

Gissio, 2025-02-01:
Converting to a discussion as this is not an issue (see #32).