Gissio/radpro #198: Bosean fs-5000 TUBE_CONVERSIONFACTOR_J321 (vs Radiacode)

PsychoBitchy, 2025-02-01:
Hello, first of all, thank you for Radpro!
I've flashed my Bosean fs-5000 with a J321 Tube with radpro.
I have a side Radiacode device for reference.
When the Bosean fs5000 is configurerd with the J321 tube setting, there is a consistent value difference with the radiacode, but also with another [FNIRSI GC-01].
The [FNIRSI GC-01] + RADPRO with the proper tube type config is somewhat consistent with the radiacode reading.
The BOSEAN fs5000 + RADPRO with the proper tube type config isn't ( ~x3 background ) = .09 µsvh vs .03 µsvh.

I get a consistent comparative reading with 400 CPS / µSv / h on my device ( not 153.8 like in settings )

https://github.com/Gissio/radpro/blob/e4900b09ae63c2ca5cd505b9d496f52740ee5a16/platform.io/src/tube.c#L148

[image]

maybe it's just mine but?

NuclearPhoenixx, 2025-02-01:
Keep in mind that the Radiacode is energy compensated, has almost non-existant response to beta and even has energy resolution to calculate exact values, while the FS-5000 does not have any of this and easily registers betas. This hugely affects the dose rate reading of course, so it always has to be taken with a grain of salt. You will never get the same values with a simple Geiger tube when compared to other better quality devices. Geiger detectors are intrinsically always only a kind of approximation of the current dose rate, they cannot do any more than that, because they're unable to differentiate between different types of radiation and particle/photon energy. I don't know about the GC-01.

Gissio, 2025-02-01:
May I suggest you read https://github.com/Gissio/radpro/blob/main/docs/field-guide.md?

Gissio, 2025-02-01:
Converting to a discussion as this is not an issue (see #32).